suaveolent/hoymiles-wifi #4: Update frequency workaround

philiip11, 2024-03-02:
I've noticed that after a restart, the limitation to two updates per minute is not enforced until sunset.

So if you want to fetch the current power every second, just call `inverter.restart()` once after sunrise.
For the rest of the day, the restriction is lifted.

suaveolent, 2024-03-03:
Thanks that is indeed a nice find. 
I will evaluate that, but probably needs to be implemented in the home assistant custom component rather than the lib.

henk23, 2024-05-06:
Hi @philiip11 Thanks for this finding. I can't seem to get it to work though.
Do you mean the `restart-dtu` command?
This command either does not work or does not lift the limitation...

closed 2024-06-21 by github-actions[bot]