suaveolent/hoymiles-wifi #6: JSON in Script

JonathanN1203, 2024-03-17:
Hi,

Unfortunately I don't have the option to format the output as Json. How does this work?

My Current code:

```from hoymiles_wifi.dtu import DTU
import asyncio

dtu = DTU("10.12.11.102")
response = asyncio.run(dtu.async_get_real_data_hms())

if response:
    print(f"DTU Response: {response}")

```

suaveolent, 2024-03-17:
The json option is only available from CLI. 

If you use the library itself, you need to parse the response as JSON yourself. 

See main.py for example how to do this.

closed 2024-03-17 by JonathanN1203

suaveolent, 2024-03-17:
@JonathanN1203 quick note: you should use the real data new function. 

The hms is deprecated and will be removed in the future.

JonathanN1203, 2024-03-17:
to you mean the function `async_get_real_data` or `async_get_real_data_new`?

suaveolent, 2024-03-17:
`async_get_real_data_new`

JonathanN1203, 2024-03-17:
ok, will the name change in future?

suaveolent, 2024-03-17:
> ok, will the name change in future?

No it will stay the same. This is the name taken directly from the Hoymiles app.

JonathanN1203, 2024-03-18:
ok, perfect