suaveolent/hoymiles-wifi #7: I cannot set power limit. I error from DTU.

GamoMan, 2024-03-31:
**Describe the bug**
I cannot set power limit. I error from DTU.
I am using DTU-Pro-S with HMS-2000-4T

**To Reproduce**
Steps to reproduce the behavior:

I use following command. I did not know what is error code mean
response = await dtu.async_get_real_data_new()

**Response**
DTU Response: dtu_sn: "412-------34"
time: 1711884638
action: 8
err_code: 7
tid: 1711884632

suaveolent, 2024-04-02:
Thanks, the request for the DTU-Pro-S might be different from the DTU-Wlite or the integrated DTU of the HMS-XXXW series.

Unfortunately, I do not have DTU-Pro, so I need some help on this one.

Are you familiar with Wireshark and can you maybe record a valid power limit request done via the app?

GamoMan, 2024-04-02:
I never use Wireshark. Can you give me the tutorial?

suaveolent, 2024-04-02:
> I never use Wireshark. Can you give me the tutorial? I can read power info from the inverter by using your lib.

I would need to investigate further. As far as I'm aware, the power limit can only be adjusted when directly connected to the inverter's Wi-Fi. However, due to encryption, decrypting network packets using Wireshark may pose a challenge for those unfamiliar with the process.

Numerous tutorials are available online for capturing and decrypting encrypted Wi-Fi network traffic, provided you have access to the password. These resources offer more comprehensive explanations than I could provide here.

I will also have a look at the source code once again to see if there is a difference for the DTU Pro.

closed 2024-06-18 by github-actions[bot]