suaveolent/hoymiles-wifi #8: Set new wifi password (extralong)

tavoc, 2024-04-06:
Hi, I do have a question for a Hoymiles HMS-800W-2T

Is it possible to set 63 Charakter long wifi passwords with this tool?
The original APP limit it to about 30. 

Thank you for your work

suaveolent, 2024-04-07:
Hi, 

I have not personally tested such long passwords but you can try: 

1. Install the tool as described
2. Connect to the Hotspot of the inverter
3. Open a command prompt
4. Issue the following command and follow the steps: 

`hoymiles-wifi--host 10.10.100.254 set-wifi`

It will also work if you already have your Inverter connected to another WiFi network, then just make sure you are on the same network and replace the ip address.

suaveolent, 2024-04-08:
@tavoc Did setting a long password work this way?

tavoc, 2024-04-08:
I have not tested it yet, as I need the inverter is not at my location. But next time I travel there I will try for sure.

s-hell, 2024-04-22:
Hi everyone,

i've tried it:

`Traceback (most recent call last):
  File "/home/shellmann/.local/bin/hoymiles-wifi", line 8, in <module>
    sys.exit(run_main())
             ^^^^^^^^^^
  File "/home/shellmann/.local/pipx/venvs/hoymiles-wifi/lib/python3.11/site-packages/hoymiles_wifi/__main__.py", line 385, in run_main
    asyncio.run(main())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 654, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/shellmann/.local/pipx/venvs/hoymiles-wifi/lib/python3.11/site-packages/hoymiles_wifi/__main__.py", line 365, in main
    response = await command_func(dtu)
               ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/shellmann/.local/pipx/venvs/hoymiles-wifi/lib/python3.11/site-packages/hoymiles_wifi/__main__.py", line 162, in async_set_wifi
    return await dtu.async_set_wifi(wifi_ssid, wifi_password)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/shellmann/.local/pipx/venvs/hoymiles-wifi/lib/python3.11/site-packages/hoymiles_wifi/dtu.py", line 217, in async_set_wifi
    request.netmode_select = NetmodeSelect.WIFI
    ^^^^^^^^^^^^^^^^^^^^^^
TypeError: 'NetmodeSelect' object cannot be interpreted as an integer`

s-hell, 2024-04-22:
Habe mir den Code mal ein wenig angeschaut.

Nachdem ich die dtu.py angepasst habe:

> request.netmode_select = 1

Hat es auch mit einem langen Passwort geklappt. Irgend etwas scheint mit der Variable Netmode.WIFI nicht zu passen.

tavoc, 2024-04-22:
Ich teste das auf alle Fälle. Geplant jst in ca 2 Wochen dahin zu fahren. Dankeschön

suaveolent, 2024-04-22:
Th

> Habe mir den Code mal ein wenig angeschaut.
> 
> Nachdem ich die dtu.py angepasst habe:
> 
> > request.netmode_select = 1
> 
> Hat es auch mit einem langen Passwort geklappt. Irgend etwas scheint mit der Variable Netmode.WIFI nicht zu passen.

Thanks, I just pushed a new version (`0.1.9`) which should fix that error.

rp42mt, 2024-06-03:
Hello @suaveolent Thanks for your efforts! The set-wifi command, sent to the hoymiles DTU (in my case: HMS1000**W**-2T), while connected the internal DTU accesspoint will start a dialogue to configure WIFI SID and password of my personal Router? Or can it only be used to set the password of the DTU integrated accesspoint?

I would like to configure wifi without creating a hoymiles cloud account. Then send MQTT to my raspberry MQTT server.

suaveolent, 2024-06-03:
It will configure the connection to your router. The internal hotspot will always be active, though.

closed 2024-07-18 by github-actions[bot]