suaveolent/hoymiles-wifi #12: no module named 'hoymiles wifi.protobuf' + frozen importlib

BassHunger, 2024-05-21:
Hi, i have installed python3 on my linuxmint and 'hoymiles-wifi-0.1.9' via setup.py aswell the protobuf-compiler.
Laptop is connected to the microinverter network, gets the following back.

Traceback (most recent call last):
  File "/usr/local/bin/hoymiles-wifi", line 33, in <module>
    sys.exit(load_entry_point('hoymiles-wifi==0.1.9', 'console_scripts', 'hoymiles-wifi')())
  File "/usr/local/bin/hoymiles-wifi", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 171, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/usr/local/lib/python3.10/dist-packages/hoymiles_wifi-0.1.9-py3.10.egg/hoymiles_wifi/__main__.py", line 14, in <module>
  File "/usr/local/lib/python3.10/dist-packages/hoymiles_wifi-0.1.9-py3.10.egg/hoymiles_wifi/dtu.py", line 37, in <module>
ModuleNotFoundError: No module named 'hoymiles_wifi.protobuf'

something with my installation is not correct. maybe someone can help me, want to connect to the microinverter data, cause Android app is not working.
thanks in advance

elafontaine, 2024-05-21:
probably the same thing as me ; 
https://github.com/suaveolent/ha-hoymiles-wifi/issues/38

The problem lies in the new changes about frozen import.  To fix, I updated my version of python...  If not possible, you'll have to dig deeper and address this with a PR for using the right annotation.

BassHunger, 2024-05-21:
thx for quick response

i have Python 3.10.12 installed; will update to latest Python 3.12.3 now and will tell the result here

BassHunger, 2024-05-21:
I gave up after 5hrs reading and trying. Too complicated to get the pyenv running. I will buy now a Tasmota Smart Plug to get my Solar Data into Home Assistant.

suaveolent, 2024-05-22:
Thanks for reporting, I will have a look at the issue.

suaveolent, 2024-06-03:
@BassHunger  I could not reproduce the issue (neither on 3.9 nor on 3.10) can you describe how you installed the package?

Did you also try installing via pip (`pip install hoymiles-wifi`) ?

elafontaine, 2024-06-03:
@BassHunger I misread your stack traces; the module that is supposed to be installed through "pip install" dependency management wasn't installed ;  **No module named 'hoymiles_wifi.protobuf'**.  If you did "pip install", then maybe you're not running the hoymile_wifi scripts from the same python interpreter, meaning the package is not installed for the python you're using.

This can happen when you changed venv by accident.  If you want to setup maybe an hour so we can troubleshoot outside my working hours, I would be glad to help :). I also want @suaveolent to be able to focus on the core and continue to maintain this package XD.

closed 2024-07-18 by github-actions[bot]

chriser-, 2024-07-31:
Just had the same issue,

```
$ python setup.py install
$ hoymiles-wifi
...
ModuleNotFoundError: No module named 'hoymiles_wifi.protobuf'
```

After checking the setup.py output closely, I found

```
package init file 'hoymiles_wifi\protobuf\__init__.py' not found (or not a regular file)
```

I just created an empty \_\_init\_\_.py inside the protobuf folder, and ran the install again, and it copied the protobuf folder as well.

```
...
creating build\bdist.win-amd64\egg\hoymiles_wifi\protobuf
copying build\lib\hoymiles_wifi\protobuf\AlarmData_pb2.py -> build\bdist.win-amd64\egg\hoymiles_wifi\protobuf
...
```

It worked afterwards.

```
$ hoymiles-wifi
usage: hoymiles-wifi [-h] --host HOST [--as-json]
...
```

```
$ python --version
Python 3.10.5

```

suaveolent, 2024-07-31:
Great, thank you for sharing. I can add an empty __inity__.py