suaveolent/hoymiles-wifi #15: Only 200 minutes of history power data

y-hoffmann, 2024-06-04:
**The bug**
When running ``hoymiles-wifi [...] app-get-hist-power`` in the terminal or ``async_app_get_hist_power()`` in code, I only see power values for the first 200 minutes of the day (device is HMS-800W-2T)

**Expected behavior**
I assume this should be able to fetch all data for the current day.

**Where I tested this:**
- Linux Desktop
- Termux (Android)

Everything else seems to work as expected.

suaveolent, 2024-06-05:
Thanks, I'm still trying to figure out the command.

You can get different results by setting `control_point` (although that is a guess from my side, the original parameter is `cp` and `requested_day`).

I'll keep you updated once I have figured out what those do.

e.g. 

```
    async def async_app_get_hist_power(
        self,
    ) -> AppGetHistPower_pb2.AppGetHistPowerResDTO | None:
        """Get historical power."""

        request = AppGetHistPower_pb2.AppGetHistPowerResDTO()
        request.control_point = 0
        request.offset = OFFSET
        request.requested_time = int(time.time())
        request.requested_day = 0
        command = CMD_APP_GET_HIST_POWER_RES
        return await self.async_send_request(
            command,
            request,
            AppGetHistPower_pb2.AppGetHistPowerReqDTO,
        )
```

y-hoffmann, 2024-06-05:
Yeah, that works, thanks. Considering how this functions, maybe the ``cp`` is short for "chunk parameter" or something but idk.

Do you know if the request return is somehow always limited to 200 data points? Using multiple requests with different ``cp`` values could be the intended way to access the history data.

suaveolent, 2024-06-05:
It seems the response is always limited to 200 data points. At least I could not find a parameter which would increase that.

closed 2024-07-20 by github-actions[bot]

leneffets, 2025-03-26:
Are here any news on that? I stumbled on the same issue. Is there a param to pass control_point?

y-hoffmann, 2025-03-26:
I used to manually change it in [this line](https://github.com/suaveolent/hoymiles-wifi/blob/main/hoymiles_wifi/dtu.py#L201) for testing only because I ended up not really needing power data.

Maybe there is a way to do this for power data similarly to how it is done [here](https://github.com/suaveolent/hoymiles-wifi/blob/main/hoymiles_wifi/dtu.py#L145), but I would wait for the author's reponse. There at least does seem to be this [`access_point` field](https://github.com/suaveolent/hoymiles-wifi/blob/main/hoymiles_wifi/protobuf/AppGetHistPower.proto#L12) (or `ap` I assume) in the response from the function [`async_app_get_hist_power()`](https://github.com/suaveolent/hoymiles-wifi/blob/main/hoymiles_wifi/dtu.py#L195C15-L195C39).