suaveolent/hoymiles-wifi #17: HMT 2250 6T - No AC Power Data

Sonnenwerk, 2024-06-12:
With my DTU/Device Combo, i get the DC information for each port, but the AC values and the temp and some others are missing.
Any input i can give you to add this information ?

suaveolent, 2024-06-12:
Thanks for reporting.

Can you maybe post the output from the `get-real-data-new` and `get-real-data` commands from the `hoymiles-wifi` lib?  https://github.com/suaveolent/hoymiles-wifi

Please only keep the first 4 digits of the inverter serial number.

Sonnenwerk, 2024-06-12:
Get-real-data Response: 
dtu_sn: "4130A003D1AB"
timestamp: 1718219437
device_number: 1
package_number: 1
pv_data {
  pv_sn: 21451488897285
  pv_port: 1
  pv_vol: 2293
  pv_cur: 2308
  pv_power: 2285
  pv_energy_total: 3983
  grid_vol: 3978
  grid_vol_max: 3968
  grid_freq: 5000
  pv_fault_num: 253
  pv_fault_cnt: 3
  pv_link_status: 1
  pv_send_power: 100
}

Get-real-data-new Response: 
device_serial_number: "4130A003D1AB"
timestamp: 1718219481
active_power: 1
firmware_version: 1
tgs_data {
  serial_number: 21451488897285
  firmware_version: 1
  voltage_phase_A: 2300
  voltage_phase_B: 2302
  voltage_phase_C: 2290
  voltage_line_AB: 3988
  voltage_line_BC: 3980
  voltage_line_CA: 3969
  frequency: 4999
  temperature: 252
  warning_number: 3
  link_status: 1
  modulation_index_signal: 100
}
pv_data {
  serial_number: 21451488897285
  port_number: 1
  voltage: 394
  power: 2
  energy_total: 73438
  energy_daily: 1014
  error_code: 50331648
}
pv_data {
  serial_number: 21451488897285
  port_number: 2
  voltage: 394
  power: 1
  energy_total: 256
  energy_daily: 4
  error_code: 50331648
}
pv_data {
  serial_number: 21451488897285
  port_number: 3
  voltage: 403
  current: 1
  power: 5
  energy_total: 115204
  energy_daily: 1730
  error_code: 50331648
}
pv_data {
  serial_number: 21451488897285
  port_number: 4
  voltage: 403
  current: 1
  power: 5
  energy_total: 118009
  port_number: 4
  voltage: 403
  current: 1
  power: 5
  energy_total: 118009
  energy_daily: 1809
  error_code: 50331648
}
pv_data {
  serial_number: 21451488897285
  port_number: 5
  voltage: 402
  current: 1
  power: 4
  energy_total: 137607
  energy_daily: 1946
  error_code: 50331648
}
pv_data {
  serial_number: 21451488897285
  port_number: 6
  voltage: 402
  current: 1
  power: 6
  energy_total: 142208
  energy_daily: 2059
  error_code: 50331648
}

suaveolent, 2024-06-13:
Alright got it.

This is a three phase inverter. I need to refactor the code to suppor these two. Might take some time though.

If you want to manually read them in the mean time you can change line `83` of the `sensors.py` from `        key="sgs_data[<inverter_count>].active_power",`  to `"active_power",`

and line `124` from `key="sgs_data[<inverter_count>].temperature",` to `tgs_data[<inverter_count>].temperature`.

If you want to read voltage too, you need to duplicate line `131-137`  for each of the voltage lines (e.g. `voltage_phase_A`

suaveolent, 2024-06-14:
@Sonnenwerk  I just pushed support for three phase inverters. You can try it by updating to `main` branch.

You need to delete the device config first and then set up the integration again for it to work properly as I did not implement any migration functions.

Please let me know if it works as intended :)

Sonnenwerk, 2024-06-14:
I deleted the devices and the ip-adress. Then i did a restart..but then..i guess i did not the right things.
Currently when i install pip install hoymiles-wifi i get version 0.2.1. Github says latest release 0.2.3. 
So i uninstalled with pip uninstall ... and reinstalled. 

When i go to add ons now, hoymiles does not start. When i want to readd the repository (i dont know where to remove it so) within Hacs add rep..and this url https://github.com/suaveolent/ha-hoymiles-wifi i get an "https://github.com/suaveolent/ha-hoymiles-wifi is not a valid add-on repository". I guess, i messed it up.

suaveolent, 2024-06-14:
Yes it seems right, but what you need to do after install is „upgrade“ to main branch since I did not yet release a new version. 

So: 

1. Remove configuration
2. Go to HACS and redownload Hoymiles but select „main“ as version
3. Configure your DTU again

Sonnenwerk, 2024-06-14:
Hmm..that update worked..but i just can see one DTU now and no inverter..is this maybe related to "night". I will check tomorrow.

suaveolent, 2024-06-15:
You are right, I experience a similar behavior. I will push a new version tomorrow. Maybe this will help.

suaveolent, 2024-06-16:
@Sonnenwerk I just pushed a new pre-release version `0.2.4-dev`.

To install it proceed as follows:

1. Delete any existing configuration
2. Navigate to HACS
3. Click on "Redownload" for the Hoymiles integration.
4. Enable "Show beta versions" and select `0.2.4-dev`
5. Restart Home Assistant
6. Setup your DTU

Now your inverter should show up.

Sonnenwerk, 2024-06-16:
Installation works fine now.
Results are as follows:
1. My HMS800W-2T is identified as a HMS1000-2T
2. The name of HMT2250-6T is correct
3. The Model information from DTU of HMS is now displayed with it´s name "DTUBI..." instead of it´s model type (integrated or whatever)
4. The HMS DTU has a value for "AC daily total" the HMT field for it is "unknown"
5. The HMT itselfs shows a lot of different values (temp/version/DC), but no AC values at all (despite net frequency).

suaveolent, 2024-06-16:
Thanks for testing. 

1. Can you maybe send me the first 6 digits of your serial number?
2. Great :) 
3. DTUBI is expected since this means the integrated DTU.
4. It seems like the DTU does not report the AC value. I need to look for another way. 
5. yes this is expected since the DTU does not report it for the Inverter.

Sonnenwerk, 2024-06-16:
1. 141293

suaveolent, 2024-06-16:
Alright that’s a bummer. This means I cannot distinguish between 800W and 1000W. 

I need to combine them as 800W/1000W

Sonnenwerk, 2024-06-26:
Can you look into those values, please ?

pv_vol: 2293
pv_cur: 2308
pv_power: 2285
pv_energy_total: 3983

I only get one of those values. I guess it is pv_cur.

suaveolent, 2024-09-13:
> Can you look into those values, please ?
> 
> pv_vol: 2293 pv_cur: 2308 pv_power: 2285 pv_energy_total: 3983
> 
> I only get one of those values. I guess it is pv_cur.

Yes, it seems that your DTU reports more values than the HMS-XXXXW series. I will need some time to implement a mechanism to correctly display it if available.

Sonnenwerk, 2024-09-13:
Just contact me when you want to continue..

suaveolent, 2024-11-05:
@Sonnenwerk I noticed that you issued the "real data" and not `real-data-new` request. The values mentioned by you should already be provided, namely:

`voltage`, `current`, `power` and `energy_total`. 

Is there anything else you are missing?

Sonnenwerk, 2024-11-25:
Good morning, yes i still miss the "AC total energy daily" value. I can not display it, nor can i find it in the data.
[image]
[image]
[image]

This time i tried the data_new, but i can not see the AC daily total power there.

suaveolent, 2024-11-25:
Unforetunately the DTU-W100/WLite does not provide the daily_energy on DTU basis.

I removed that sensor for these DTUs. However, you still need to manually delete it from your DTU in Home assistant.

Sonnenwerk, 2024-11-25:
Oh, that is sad. How does the hoymiles cloud identify that information ?

suaveolent, 2024-11-25:
I assume they calculate the value based on the `energy_daily`. You could do that also with a helper in home assistant.

Sonnenwerk, 2024-11-25:
I made myself a not as workaround todo this. But i stumbled a little bit, that those values are the DC Input, not AC output. So i assume a pure sum will not fit.

suaveolent, 2024-11-25:
Unfortunately not. I am already reading these values for each inverter.

For the DTU energy (all ports combined) you would need a value `dtu_daily_energy` which apparently the DTU WLite does not supply.

Sonnenwerk, 2024-11-25:
Ok. Thank you for looking into it !

Sonnenwerk, 2024-11-25:
.

closed 2024-11-25 by Sonnenwerk