suaveolent/hoymiles-wifi #18: Response is always None

MaierJuerg, 2024-06-22:
Created a new project with python3.11

This is  my code:
```
from hoymiles_wifi.dtu import DTU
import asyncio

dtu = DTU("192.168.0.19")

async def main():
    response = await dtu.async_network_info()
    if response:
        print(f"DTU Response: {response}")
    else:
        print("Unable to get response!")

asyncio.run(main())

```

**Expected behavior**
return information, response however is always none, also when trying other functions

**Desktop (please complete the following information):**
ubuntu 23.10
pycharm 2024.1.1, community edition

**Additional context**
I run the code on a laptop.
I have a Hoymiles HMS-800-2T device
My WiFi is connected to the DTU (DTUBI-<SerialN>
I have set the IP manually to 192.168.0.19 and I can ping it successfully
I can see my plant in the S-Miles-Cloud browser
The only thing might be that the browser shows my plant as "offline" and I can't see how to bring it online

elafontaine, 2024-06-22:
Ok, the hoymile_wifi is using the "wifi" port of your Wifi router (not the DTU's Wifi).   You can test this by doing a TCP connection of any kind (telnet) to port 10081.

MaierJuerg, 2024-06-23:
Thanks, not so sure I can follow your response.
I run a ifconfig on my laptop and it shows me a wlp2s0 device with IP 192.168.0.108
You say I should enter this address as the DTU-IP in my python script?
With this I still get "unable to get response"
With  my phone(Android) in the s-miles installer app I can get a DTU-Router connection when activating the hotspot in the phone. It can however not connect to the server. Selecting the ToolKit in the app I get power production details shown.
My versions:
DTU HW H00.01.00, DTU SW: V00.01.11
Micro-SN H00.04.00, Micro-SN SW: V01.00.08

MaierJuerg, 2024-06-23:
Can't understand how your code works. Where might I find the logger file and what is the filename?

elafontaine, 2024-06-23:
Did you connect the DTU to your home wifi for the data transmission to s-miles ?

If yes, then connect your laptop to that wifi network, not the DTU wi-fi.

In my example, the AT&T provided wifi router is the network my laptop AND DTU are connected to.  The DTU used DHCP to get it's IP address on my home network.   

You mention that you have to activate your phone wifi hotspot (and not just the wifi network)... I'm wondering something.  Are you able to see your data in s-miles WITHOUT having your phone hotspot active ?  

I'll try to provide a drawing to share what I mean.  Because the documentation of hoymiles isn't clear in the first place... and this is an undocumented port...

MaierJuerg, 2024-06-23:
Sorry, i had expected the setup to be an easy walk. I have tried so many things, can't remember all the steps I tried. Maybe remove my "plant" from the s-miles-cloud and create a new one following exactly the pdf instructions?
On my phone:
Switched Hotspot off, WLAN is set to DTUBIxxx and with the toolbox I can retrieve data.
When leaving toolkit it shows me a message that I need to diconnect with the DTU to use the internet
the DTUBI Wifi connection is set to DHCP and gets a 10.10.100.150 IP address (I assume from the phone)
I can not see how the DTU device can connect to the internet (WiFi connection status is always shown as "no internet")
Maybe this thing is not designed to work for network-idiots :-(
I can also not understand why they have created an installer and end user app. the end user app always tells me to use the "installer app"?

elafontaine, 2024-06-23:
There is multiple way to setup the DTU, but they're all similar.

1- the 1-Sub-G interface that is made to connect to the micro-inverters.  This is always setup and available.

2- the DTU Wi-Fi network.  This is the network that you can connect to to configure the DTU from your phone (that's the one the toolkit needs to use, the DTUBI with network range of 10.10.100.x/24 in your case)

3- The Ethernet cable or Wifi network connectivity.  This is what is used to send data from the DTU to the S-Miles cloud where you can retrieve the data of your panels, DTU and micro-inverters.  If you're using 'wifi', then there is a port exposed (tcp port 10081) that this library can use on this network.  But your DTU needs to have been configured to use the wifi network to send it's data.  It's not the DTUBI network.  

Notice that the connectivity described in number 3 is not set up by the DTU.  It's something the DTU use as a client.  This means it will use DHCP to get an IP that you will need to figure out to use this library.  Normally, internet provider router expose an internal "website" to allow you to configure your router and retrieve information (like how many devices are connected and what are the IP addresses that they use).

At this point, I can't help more than that... the drawing will just highlight what I've described right now.  

You shouldn't have to redo a plant... but the plant should see data.  If you do not see data in the s-miles cloud application WITHOUT the phone app, then it means your base setup isn't done right.  You need the setup dine right before being able to use this library.  Also, use the webpage to upgrade the version  of the DTU.  They don't ship with the latest (I've had my setup done in May and had to upgrade).

elafontaine, 2024-06-23:
What is your DTU model ?

MaierJuerg, 2024-06-23:
It is a dhs-800-t2 as of my knowledge - but I can not query it :-(

elafontaine, 2024-06-23:
That would be the inverter, but what about the DTU itself ? e.g. I got the DTU-Pro-S from hoymiles.

see https://www.hoymiles.com/products/product-list#GHz

MaierJuerg, 2024-06-24:
I only have the inverter. Do I need to purchase a DTU to make things work?
My setup delivers energy (I have a tasmota smart switch to see that).

MaierJuerg, 2024-06-24:
> No your setup should be fine If you have a wifi enabled inverter.
> Contrary to what @elafontaine <https://github.com/elafontaine> said, you
> can either directly connect to the DTU or connect to it via your wifi
> router, when you already have successfully set it up.
> If you directly connect to the DTU, you should use 10.10.100.254.
> Otherwise you need to use the IP provided by your router.
> Since you cannot see the inverter in the s cloud, I suspsect that there is
> an issue with your connection to the router.
> Try updating the firmware and make sure that your inverter can reach the

thanks suaveolent
still struggling...
I see the DTU in my laptops wifi settings. I can connect to it, it shows
(Connected)
In the settings I have assigned it a static ip 192.168.0.200 and I can ping
it
running the python script however still returns none.
I have been able to activate logging and it shows:

DEBUG:hoymiles_wifi:hoymilesWifi start
DEBUG:asyncio:Using selector: EpollSelector
DEBUG:hoymiles_wifi:[Errno 111] Connect call failed ('192.168.0.200', 10081)
DEBUG:hoymiles_wifi:DTU is NetworkState.Offline
DEBUG:hoymiles_wifi:Error closing writer: cannot access local variable
'writer' where it is not associated with a value

could NetworkState.Offline be the problem and how might I bring it Online?
Regards
Juerg

elafontaine, 2024-06-24:
> Contrary to what @elafontaine said, you can either directly connect to the DTU or connect to it via your wifi router, when you already have successfully set it up.
I admit I was wrong, just tested it and the IP I had to reach was 10.10.100.254.  I found it because it was the "default gateway" given by the DHCP server (which was also 10.10.100.254). 

If for some reason the default value changed for you, you could list who is the DHCP server attributing the device an IP address on that network.

May  I suggest you run a simple 

> No your setup should be fine If you have a wifi enabled inverter.

I see I was wrong there as well ; https://www.hoymiles.com/wp-content/uploads/2023/06/Datasheet_HMS-600W700W800W900W1000W-2T_EU_EN_V202305.pdf

I was using the other HMS-800W-2t model which doesn't have a Wi-fi integrated.

elafontaine, 2024-06-24:
Have you followed this guide ? 
https://www.hoymiles.com/wp-content/uploads/2024/01/Quick-Installation-Guide_HMS-1000W-Series_EU_EN_REV1.1.pdf

Because at the last page, the method 2 is what I was expecting you wanted ; It would sent the data to hoymiles and your plant should come online then.

MaierJuerg, 2024-06-24:
I retried a few things. In the s-miles-installer-app in Network Config I
first used my phones hotspot wifi.
I switched this to my home-mesh-wifi (that can see the DTU) and it started
to work, first connecting to the router, than the server. My plant is
online now.
So I tried hoymiles-wifi again but no luck. No idea what IP I need to
enter. The wifi is a Honor3 mesh that uses its own IP-range. I can see the
DTU and IP in the Honor AI-Life app 192.168.7.72 but cannot query it (my
local ip range is 192.168.0.x)
My main router does not list devices connected through the mesh.

elafontaine, 2024-06-24:
Good :) now we can start playing around the network you've got setup.

The first part to understand is "who is serving 192.168.7.0/24?", is this the Honor 3 router you mention ? You mention your laptop is in the range of 192.168.0.0/24, which I'm guessing the network 192.168.7.0/24 is nated behind a network's 192.168.0.0/24 address.  Otherwise, your laptop should have reached it with the 192.168.7.72 address if it was the reverse.  

One easy solution would be to connect your laptop to the Honor 3 router (https://www.honor.com/global/routers/honor-router-3/ this one ?).  This would allow your laptop to still have internet access and also be in the same network as the "DTU", so it will be a direct communication.  I do not understand if you have a reason why you didn't use the same network for it though.   Even if just temporary, I would try it out to confirm port 10081 is open on the DTU.

Another solution would be to go into the Honor3 router and open port 10081 toward the DTU port 10081.  That way your laptop could reach the **Honor3 router ip (WAN side)** to reach onto the DTU behind it (LAN side).  

At this point, it's networking ;) the products seems to work as expected as you're able to see it in the App on your browser.  I do recommend to update the firmware version right away though.  Many bugs I had were fixed that way.

Let us know if you get it to work :)

MaierJuerg, 2024-06-25:
Hi Eric
Your knowledge with networks looks to be much better then mine. Let me at
least try to answer your questions

*"The first part to understand is "who is serving 192.168.7.0/24
<http://192.168.7.0/24>?", is this the Honor 3 router you mention ?"*
yes, it is a mesh of 4 routers covering house and garden area. I initially
had my main router on 192.168.0.0/2 <http://192.168.0.0/24>4
 and saw, that the mesh used 192.168.1.0/24. I thought this to be a bit
strange because I wanted to see all my connected devices in 192.168.0.0/24,
When I set my main routers range to 192.168.0.0/2 <http://192.168.0.0/24>2
to include the 92.168.1.0 <http://192.168.0.0/24> range the mesh simply
switched its range to 192.168.7.0/24 <http://192.168.0.0/24> and I ended up
in the same situation.
I am using a few ESP devices with temp/humidity sensors, 2 ESP32cam and  my
tasmota smart switches and they all get 192.168.7.x addresses. I have a
mqtt server in my local network and can exchange messages with all my
devices, so I stopped bothering about this.

*" You mention your laptop is in the range of 192.168.0.0/24
<http://192.168.0.0/24>,"*
Almost right, the laptop is almost always wire connected with my main
router and has the fixed IP 192.168.0.12. The main router is currently set
to 192.168.0.0/22 <http://192.168.0.0/24> (subnet mask 255.255.252.0)
ifconfig: inet 192.168.0.12  netmask 255.255.252.0  broadcast 192.168.3.255

*"which I'm guessing the network 192.168.7.0/24 <http://192.168.7.0/24> is
nated behind a network's 192.168.0.0/24 <http://192.168.0.0/24> address.
Otherwise, your laptop should have reached it with the 192.168.7.72 address
if it was the reverse."*
Right, my laptop without activated wifi can not see the 192.168.7.x
addresses.

*"One easy solution would be to connect your laptop to the Honor 3 router
(https://www.honor.com/global/routers/honor-router-3/
<https://www.honor.com/global/routers/honor-router-3/> this one ?). This
would allow your laptop to still have internet access and also be in the
same network as the "DTU", so it will be a direct communication."*
It is the router model I am using, yes.
Followed your suggestion and added the mesh wifi to my laptop wifi settings.
Tadaaa - I can ping my DTU 192.168.7.78 and running the hoymiles-wifi
script can capture information.

" I do not understand if you have a reason why you didn't use the same
network for it though. Even if just temporary, I would try it out to
confirm port 10081 is open on the DTU."
I assume port 10081 is open as I can get information from the device?

"Another solution would be to go into the Honor3 router and open port 10081
toward the DTU port 10081. That way your laptop could reach the *Honor3
router ip (WAN side)* to reach onto the DTU behind it (LAN side)."
that actually would be the best solution to me as my (self designed) python
home automation machine sits in the basement and has only week access to
the wifi mesh.
The task looks a bit above my head now but I can start to investigate.
I have in the past wondered how to close the gap between my 192.168.0 and
192.168.7 ranges. Something like a software-bridge to combine them?

First task will be to extend the hoymiles-wifi script to create mqtt
messages about my energy-harvesting status, add it to my data-capturing
task and extend my dash app to show the progress. I would much prefer to
get rid of any cloud dependencies of my devices as I wasted a lot of time
with Meross and Tuya cloud integration into my app.

Thanks a lot for your patience and helpful comments.
Regards
Juerg

elafontaine, 2024-06-26:
Pleasure :) and yes, the CIDR 192.168.0.0/22 does not encapsulate the 192.162.7.0 range.

So my guess that those are 2 distinct network is right.

you did confirm that the port 10081 was open on the DTU, with your test on the mesh wifi network.  Now, it's just a question of figuring out how to configure your Honor 3 router to "serve" request on port 10081 (but this will actually send them to the DTU).  This is called port forwarding and should be available in most routers.

I do not know your setup, but here is what I assume is going on; 
- you probably have a ISP provided router that gives you connectivity to internet (192.168.0.0/22).  This is where your laptop is connected and where all the requests going to internet are nated (acting as if it was the router that made the request on internet).  
- I also assume that your honor3 router  WAN port  is connected on the same ISP provided network and probably getting an IP from the ISP router on it's LAN (Local area network).  
- Then your mesh network is made available from your honor 3 router Wi-fi network.

If all my assumptions are true, then my proposed solution to you is port forwarding.

You could make another setup that would basically make it all a single network but the configuration for it would be complicated to someone that doesn't understand DHCP and DNS configuration (and bridging).

In the mean time, if you have questions, you can ask here.

Were you planning on using Home-Assistant ? if yes, @suaveolent already made a package on HACS to integrate this package into HA more easily.

MaierJuerg, 2024-06-27:
Hi Eric
I was able to relocate my basement laptop to have a better connection with
the mesh wlan. So I am able to run the hoymiles-wifi script as part of my
data capturing to persist the power production values in my own data files.
I have also seen that there is an arduino solution to access the DTU. So I
could setup an ESP8266 to query the data from the DTU and send it to my
mqtt broker.
The router features of the Honor3 look rather restricted to me and I doubt
I could set up a port forwarding.
I run my own python data capturing and visualization app, not a commercial
ha solution.
Thanks again for your support
Regards
Juerg

closed 2024-08-11 by github-actions[bot]