suaveolent/hoymiles-wifi #19: Error output as json with commands identify-dtu and identify-inverters

pipp37, 2024-06-25:
**Thanks**
First thanks for great tool. 
I made a docker container and found a issue on 2 commands with json output.

**Describe the bug**
Output as json generates some errors with hms800w-2t and DTUL. 
Normal output runs fine with Response: 

Coommands: 
- identify-dtu 
- identify-inverters

**To Reproduce**
Steps to reproduce the behavior:
1. RUN: hoymiles-wifi --host 192.168.28.193  identify-inverters - OK
2. RUN: hoymiles-wifi --host 192.168.28.193  identify-inverters --as-json  - ERROR
3. See error

**Expected behavior**
Output should generate json string

**Screenshots**
look at attached logfile.

**Linux :**
 - OS: debian-bullseye - docker AMD64
 - Microinverter HM800w-2t at 192.168.28.193
 - Docker container: https://hub.docker.com/r/pipp37/hoymiles-wifi-python/tags
 - CMD:  docker run -it --rm pipp37/hoymiles-wifi-python:1.0 bash
 - hoymiles-wifi --host 192.168.28.193  identify-dtu --as-json  (Error)
 - hoymiles-wifi --host 192.168.28.193  identify-dtu  (OK)
 - Version hoymiles-wifi: 0.2.2

[bash-shell-log-error.txt](https://github.com/user-attachments/files/15971123/bash-shell-log-error.txt)

pipp37, 2024-06-25:
Docker python container: python:3.12.4-slim-bullseye

suaveolent, 2024-07-26:
Thanks I somehow missed this issue: will have a look at it.

commit 07ee0b91 (2024-09-03): fix as json for identitfy requests, fixes #19
commit be21769e (2024-09-13): fix as json for identitfy requests, fixes #19
closed 2024-09-03 by suaveolent