suaveolent/hoymiles-wifi #20: Difference of "Energy Total" to Hoymiles "App Energy" total

sarns, 2024-07-11:
Hello,

first of all thanks a lot for providing this library! I use it on my Raspberry Pi to poll the data each minute and write it to an Influx DB. After some days of recording, I just recognized that there is a difference between the values I get from the Hoymiles App and this library for Energy total.

`Panel1 E-total App:         4.92 kWh`
`Panel1 E-total Influx:      6.79 kWh`
`Panel2 E-total App:        10.59 kWh`
`Panel2 E-total Influx:     11.30 kWh`

Can somebody reproduce that?

suaveolent, 2024-07-11:
This looks like a delay to me: the App pulls data from the cloud and depends on the inverter sending these data. However, this is only done every 15 minutes or so.

suaveolent, 2024-07-11:
Might be also due to the fact that the PV data total is DC, the daily value is AC. 

So you have some kind of loss there.

closed 2024-08-25 by github-actions[bot]