suaveolent/hoymiles-wifi #21: Strange values for HMS-700-2T inverters

tannercollin, 2024-08-11:
I'm getting strange values for HMS-700-2T inverters. For example: 2496 voltage, 6000 frequency, and active power being way higher than what the array is spec'd for. Nominal array power is 410 W + 23 = 9430 W.

Not sure what's happening, I suspect the protobuf fields don't align? I tried poking around in `RealDataNew.proto`, but couldn't make any sense of it. Do you have any pointers? Thanks!

Here's my get-real-data response:

```
Get-real-data Response: 
dtu_sn: "4121A00*****"
timestamp: 1723399229
device_number: 2
package_number: 1
rp_data {
  rp_sn: 189950310*****
  rp_signal: 1
  rp_channel: 2495
  rp_link_nub: 5998
  rp_link_status: 4554
}
rp_data {
  rp_sn: 189950310*****
  rp_signal: 1
  rp_channel: 2496
  rp_link_nub: 5998
  rp_link_status: 4563
}
rp_data {
  rp_sn: 189950310*****
  rp_signal: 1
  rp_channel: 2498
  rp_link_nub: 5998
  rp_link_status: 2250
}
rp_data {
  rp_sn: 189950310*****
  rp_signal: 1
  rp_channel: 2497
  rp_link_nub: 5998
  rp_link_status: 4523
}
rp_data {
  rp_sn: 189950310*****
  rp_signal: 1
  rp_channel: 2502
  rp_link_nub: 5998
  rp_link_status: 5547
}
rp_data {
  rp_sn: 189950310*****
  rp_signal: 1
  rp_channel: 2502
  rp_link_nub: 5999
  rp_link_status: 6141
}
rp_data {
  rp_sn: 189950310*****
  rp_signal: 1
  rp_channel: 2496
  rp_link_nub: 5999
  rp_link_status: 6139
}
```

And here's get-real-data-new:
```
Get-real-data-new Response: 
device_serial_number: "4121A00*****"
timestamp: 1723399280
active_power: 2
firmware_version: 1
sgs_data {
  serial_number: 189950310*****
  firmware_version: 1
  voltage: 2495
  frequency: 6001
  active_power: 4547
  reactive_power: 628
  current: 182
  power_factor: 990
  temperature: 441
  warning_number: 1
  link_status: 1
  modulation_index_signal: -73
}
sgs_data {
  serial_number: 189950310*****
  firmware_version: 1
  voltage: 2497
  frequency: 6001
  active_power: 4563
  reactive_power: 630
  current: 182
  power_factor: 990
  temperature: 414
  warning_number: 1
  link_status: 1
  modulation_index_signal: -67
}
sgs_data {
  serial_number: 189950310*****
  firmware_version: 1
  voltage: 2497
  frequency: 6001
  active_power: 2248
  reactive_power: 628
  current: 90
  power_factor: 963
  temperature: 344
  warning_number: 3
  link_status: 1
  modulation_index_signal: -82
}
sgs_data {
  serial_number: 189950310*****
  firmware_version: 1
  voltage: 2495
  frequency: 6001
  active_power: 4517
  reactive_power: 628
  current: 181
  power_factor: 990
  temperature: 373
  warning_number: 2
  link_status: 1
  modulation_index_signal: -70
}
sgs_data {
  serial_number: 189950310*****
  firmware_version: 1
  voltage: 2500
  frequency: 6001
  active_power: 5548
  reactive_power: 632
  current: 221
  power_factor: 993
  temperature: 480
  warning_number: 1
  link_status: 1
  modulation_index_signal: -73
}
sgs_data {
  serial_number: 189950310*****
  firmware_version: 1
  voltage: 2501
  frequency: 6001
  active_power: 6132
  reactive_power: 631
  current: 245
  power_factor: 994
  temperature: 497
  warning_number: 2
  link_status: 1
  modulation_index_signal: -67
}
sgs_data {
  serial_number: 189950310*****
  firmware_version: 1
  voltage: 2494
  frequency: 6001
  active_power: 6134
  reactive_power: 628
  current: 245
  power_factor: 994
  temperature: 494
  warning_number: 1
  link_status: 1
  modulation_index_signal: -62
}
pv_data {
  serial_number: 189950310*****
  port_number: 1
  voltage: 291
  current: 821
  power: 2390
  energy_total: 14230
  energy_daily: 743
  error_code: 50396928
}
pv_data {
  serial_number: 189950310*****
  port_number: 2
  voltage: 292
  current: 826
  power: 2411
  energy_total: 15181
  energy_daily: 884
  error_code: 50396928
}
pv_data {
  serial_number: 189950310*****
  port_number: 1
  voltage: 296
  current: 814
  power: 2411
  energy_total: 12090
  energy_daily: 444
  error_code: 50986752
}
pv_data {
  serial_number: 189950310*****
  port_number: 2
  voltage: 294
  current: 817
  power: 2407
  energy_total: 14196
  energy_daily: 740
  error_code: 50986752
}
pv_data {
  serial_number: 189950310*****
  port_number: 1
  voltage: 2
  current: 1
  error_code: 50396928
}
pv_data {
  serial_number: 189950310*****
  port_number: 2
  voltage: 292
  current: 813
  power: 2374
  energy_total: 14813
  energy_daily: 864
  error_code: 50396928
}
pv_data {
  serial_number: 189950310*****
  port_number: 1
  voltage: 291
  current: 813
  power: 2373
  energy_total: 14855
  energy_daily: 877
  error_code: 50986752
}
pv_data {
  serial_number: 189950310*****
  port_number: 2
  voltage: 293
  current: 816
  power: 2397
  energy_total: 14816
  energy_daily: 870
  error_code: 50986752
}
pv_data {
  serial_number: 189950310*****
  port_number: 1
  voltage: 287
  current: 899
  power: 2584
  energy_total: 16973
  energy_daily: 507
  error_code: 50986752
}
pv_data {
  serial_number: 189950310*****
  port_number: 2
  voltage: 282
  current: 1160
  power: 3274
  energy_total: 18335
  energy_daily: 1022
  error_code: 50986752
}
pv_data {
  serial_number: 189950310*****
  port_number: 1
  voltage: 281
  current: 1152
  power: 3238
  energy_total: 18191
  energy_daily: 1023
  error_code: 50986752
}
pv_data {
  serial_number: 189950310*****
  port_number: 2
  voltage: 279
  current: 1160
  power: 3236
  energy_total: 18148
  energy_daily: 1029
  error_code: 50986752
}
pv_data {
  serial_number: 189950310*****
  port_number: 1
  voltage: 278
  current: 1158
  power: 3224
  energy_total: 18221
  energy_daily: 1022
  error_code: 50986752
}
pv_data {
  serial_number: 189950310*****
  port_number: 2
  voltage: 280
  current: 1161
  power: 3253
  energy_total: 18265
  energy_daily: 1025
  error_code: 50986752
}
dtu_power: 59429
dtu_daily_energy: 17078
```

suaveolent, 2024-08-11:
These data actually look fine. You need to divide by e.g. 100 to get the real value. 

Have a look at:

https://github.com/suaveolent/ha-hoymiles-wifi/blob/main/custom_components/hoymiles_wifi/sensor.py

There you can see the right conversion for each value.

tannercollin, 2024-08-11:
You're right, thank you!

closed 2024-08-11 by tannercollin