suaveolent/hoymiles-wifi #26: AC total energy

di-le, 2024-08-21:
Hello,

I have a Hoymiles HMS-800W-2T and have integrated your script into Home Assistant. This has been going on there for a few days now. Thank you for that.

But now I noticed that I have so few AC values. I have a lot of DC values ​​but only two AC values. 
AC daily energy and AC performance.

But I miss an AC total energy. In the app without the cloud, I can also see all the energy since the system was in operation.
Is such a value actually not delivered via this route or am I missing something?

Thank you very much greetings

[image]

suaveolent, 2024-09-03:
Unfortunately there is no AC total energy. You need to set that up using a Riemann sum integral in home assistant yourself.

closed 2024-09-03 by suaveolent

stevleibelt, 2024-09-04:
@di-le if you understand @suaveolent answer, could you be so kind an share your "Riemann sum integral" as a howto with the community?

That would be great.