suaveolent/hoymiles-wifi #27: Two WLAN-Devices and two HMS-800W-2T on one Raspi

Jo-Bl, 2024-09-09:
I have two WLAN-Devices on one Raspi: the internal device and the other one is an USB-Stick, which is native supported by the RaspiOS. Both devices works well with RaspiOS as wlan0 and wlan1. 

I can connect one of the devices to the AP of HMS-800W-2T#1 and the other device to the AP of HMS-800W-2T#2. This works well. 

The wlan0 device get the ip 10.10.100.151, the wlan1 device get the ip 10.10.100.152, This works well.

But both AP of HMS-800W-2T have the same server-ip 10.10.100.254. So it is not possible to use the --host IP parameter of hoymiles-wifi alone to sepatate the micro-inverters. 

In this case it is necessary to also name the interface that is to be addressed: wlan0 wlan1

Please add this if possible. (--device or so...)

Thannks for your work.

Please note: I do not use the hoymiles-cloud and do not want to change the micro-inverters mode from Ap to part of my network.

suaveolent, 2024-09-09:
Added with `v0.2.4`. You should now be able to use `--local_addr` parameter to bind to the specific interface.

e.g.
 
`hoymiles_wifi --host 10.10.100.254 --local_addr 10.10.100.151 get-real-data-new`
`hoymiles_wifi --host 10.10.100.254 --local_addr 10.10.100.152 get-real-data-new`

closed 2024-09-09 by suaveolent

Jo-Bl, 2024-09-12:
Thanks.

But sorry, this doesn't work. Independent from --local_addr it requests server on wlan1 (10.10.100.151 -> 10.10.100.254) but sadly never wlan0 (10.10.100.150 -> 10.10.100.254)