suaveolent/hoymiles-wifi #30: S-Miles Cloud not updated when DTU is polled

mirco-frenzel, 2024-09-18:
Hi,

I`m not sure it this is a bug or just coincidence...
I have an app thats polling my DTU with "get-real-date-new" and parses the output. It`s polling every 60s, and that works as expected.
Doing this kind of prevents the DTU from reporting to the S-Miles Cloud, there it`s shown as not producing anything. 
When I enter the DTU`s own wifi (by "toolkit" screen in the s-miles app), it reflects the readings the hoymiles-wifi script is giving.

Is this a known sideeffect/itnterference with the DTUs reporting to the cloud? Maybe once a minute is too often?

Thx!

suaveolent, 2024-09-18:
As far as I know the DTU sends the data every 30 seconds or so. If you try to query the DTU during this time frame no data will be send to the cloud. 

You can try, e.g. 35 seconds and it should work just fine.

mirco-frenzel, 2024-09-18:
Ok I`ll try to experiment with the polling frequency.
I also tried once every second and got reliable readings every second, but totally blocked cloud transfer of course.
I could network-sniff the dtu to see it`s actions when it`s undisturbed...

yeckel, 2024-10-15:
Hi @mirco-frenzel, what is your refresh frequency now? I'm trying with 5s, and the power value is constant 0W from the sunrise to the sunset. The cloud is empty as well. However, when I connect with the S-Miles installer app, then I'm getting proper data till sunset.

closed 2024-11-29 by github-actions[bot]