suaveolent/hoymiles-wifi #31: Mismatch in version

Sonnenwerk, 2024-09-29:
Hi,
i have those informations available.

software version of Hoymiles integration
[image]

DTUBI aka HMS800W
[image]

DTU aka Lite-S
[image]

Looks like the Lite-S did not update correctly, or is this a mismatch with the software versioning ?

btw my HMS still works (after update), but my HMT does not display any data no more. Not sure, what i can provide that you could have a look into the issue

suaveolent, 2024-10-02:
Thanks. 

the version looks fine to me. The SW version is the version of the DTU/inverter. Not the integration. 

Regarding the missing data from HMT: can you look into the logs if there is anything? 

Otherwise you can try to reconfigure the integration by clicking on the three vertical dots next to the DTU. 

Just make sure that the DTU and inverter is available.

Sonnenwerk, 2024-10-03:
Hi,
i had a look into the data the HMT 2250 delivers and marked my suggestions. 
Regarding the "Zählermodel" - it seems to be not a SW Version, nor the Gridprofile. It seems to be something different, that is not shown in the hoymiles cloud.

device_serial_number: "4ddd"
timestamp: 172dfafaf
active_power: 1
firmware_version: 1
tgs_data {
  serial_number: dddd
  firmware_version: 1
  
  
  **active_power: 7243** --> most important information is not shown in HA
  current_phase_A: 105 --> would be nice to have it shown too
  current_phase_B: 105 --> would be nice to have it shown too
  current_phase_C: 105 --> would be nice to have it shown too
  power_factor: 1000 --> would be nice to have it shown too
  temperature: 246
  warning_number: 1
  link_status: 1
  modulation_index_signal: -82
}

}
pv_data {
  serial_number:dd
  port_number: 3
  voltage: 433
  current: 294
  power: 1276
  energy_total: 273140
  energy_daily: 153
  **error_code: 50986752** --> i guess this is a "no error"-code. It would be nice to have it also in the entities available, as it shows on the panel level if there is an error.

suaveolent, 2024-10-04:
Thanks, I should be able to integrate them for three-phase inverters.

suaveolent, 2024-10-17:
@Sonnenwerk all of the values should be added now with `v0.3.1`.

closed 2024-10-17 by suaveolent

Sonnenwerk, 2024-10-19:
I installed and checked the new version. 
The error code and active power works.
The fields for phase current are shown, but contain no value im HA
The daily energy still does not work. It should come from DTU, but it shows 0. 

[image]
[image]