suaveolent/hoymiles-wifi #34: HMS-800W-2T AC Daily power missing

mrk-gda, 2024-11-06:
Hy
There is no info of daily energy AC. There is no info total Enrgy AC as well. Is possible to fix it. Daily energy AC is most important.
Below output from inverter. dtu_daily_energy  is not AC but DC. Is possible correct this ?

mirek@mir:~$ hoymiles-wifi --host 192.168.10.93 get-real-data-new
Get-real-data-new Response: 
device_serial_number: "414393320732"
timestamp: 1730897855
active_power: 1
firmware_version: 1
sgs_data {
  serial_number: 22070011496242
  firmware_version: 1
  voltage: 2334
  frequency: 5002
  active_power: 342
  current: 14
  power_factor: 1000
  temperature: 113
  warning_number: 7
  crc_checksum: 39753
  link_status: 1
  modulation_index_signal: 12517507
}
pv_data {
  serial_number: 22070011496242
  port_number: 1
  voltage: 338
  current: 106
  power: 361
  energy_total: 16107
  energy_daily: 272
  error_code: 50921472
}
pv_data {
  serial_number: 22070011496242
  port_number: 2
  voltage: 6
  error_code: 50331648
}
dtu_power: 342
dtu_daily_energy: 272

suaveolent, 2024-11-06:
AC Daily energy should be found in the DTU device, not the inverter.

There is no total AC energy provided, unfortunately. You'll have to calculate that yourself via a helper.

mrk-gda, 2024-11-06:
This code talks to DTU integrated in HMS-800W-2t inverter.  There is data dtu_daily_energy field (last line) but data in this field is not AC energy but sum of energy from PV panels (DC energy).

suaveolent, 2024-11-06:
I'm not entirely sure if I got it right, I'll try to give an answer anyways.

From my experience,  thedtu_daily_energy is AC, since it sometimes differs from the sum of the energy_daily, especially if two modules are connected. It is usually slighty lower than the `energy_daily` field due to e.g. conversion loss.

mrk-gda, 2024-12-08:
Issue can be close. I'm not sure AC or DC daily energy production corresponds  to received  data.  I try to check it when daily production will be bigger then during winter time.

closed 2024-12-08 by suaveolent