suaveolent/hoymiles-wifi #35: Impossible to upgrade HMS-800W-2T

olivierboudet, 2024-11-09:
**Describe the bug**

The upgrade command fails

```
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! WARNING !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
!!!    Danger zone! This will update the firmeware of the DTU.     !!!
!!!  Please be careful and make sure you know what you are doing.  !!!
!!!          Only proceed if you know what you are doing.          !!!
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! WARNING !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

Do you want to continue? (y/n): y
Please select a firmware version:
1.) V00.01.11
2.) Custom URL
Enter your selection (1 or 2): 1

Firmware update URL: "http://fwupdate.hoymiles.com/cfs/bin/2311/06/,1488725943932555264.bin"

Do you want to continue? (y/n): y
No response or unable to retrieve response for firmware-update

```
But the firmware does not exists at the specified URL (404 error) : http://fwupdate.hoymiles.com/cfs/bin/2311/06/,1488725943932555264.bin 

Is there a way to find the new URL ? I didn't found anything on the hoymiles website...

Thank you

**Desktop (please complete the following information):**
 - OS: Linux
 - Browser : chromium
 - Version : 129

suaveolent, 2024-11-09:
Thanks apparently they changed the URL. I will check with my inverter.

olivierboudet, 2024-11-09:
There is a file at https://fwupdate.hoymiles.com/cfs/bin/2311/06/1488725943932555264.bin (without the comma), is this the good one ?

suaveolent, 2024-11-09:
Yes, but the comma is send to the DTU (this is okay) 

You can also check via the s Miles cloud whether you can install the update: 

https://global.hoymiles.com/

olivierboudet, 2024-11-09:
I have no account on https://global.hoymiles.com/ (and it seems we can't create one), only on S-Miles Home but the upgrade fails too. I will try to contact hoymiles support.

Thank you

suaveolent, 2024-11-09:
By the way: the firmware update command does not provide any response. You can check in home assistant whether the version upgrade was successful.

olivierboudet, 2024-11-14:
FYI the hoymiles support did the upgrade for me, they told me that firmware update is not possible for end users... You must ask them each time 🤔

suaveolent, 2024-11-14:
Yes that could be. I have an installer account so I never had that issue. 

I will close the comment  for know since I plan to evaluate the update process anyhow.