suaveolent/hoymiles-wifi #36: Home Assistant Core 2025.11

technicsdj, 2024-11-10:
Hey,

Homeassistant has the following error since the latest update. 
It should be reported to the author, since I don't see a message here, I'm doing this now. 

[image]

suaveolent, 2024-11-10:
Thanks, I will have a look at it.

suaveolent, 2024-11-10:
Fixed with `v0.3.2`

closed 2024-11-10 by suaveolent

technicsdj, 2024-11-10:
Hey,

I have updated to version 0.32 and am now getting the following error message.

Do you have any ideas?

[image]

suaveolent, 2024-11-10:
It seems like there was a problem during upgrading installing the dependencies. 

You can try installing hoymiles_wifi manually as described in the README: 

`pip install hoymiles-wifi `

technicsdj, 2024-11-10:
Unfortunately, I cannot execute a pip command via the terminal.

I have replaced the files, but unfortunately this does not help either.

suaveolent, 2024-11-11:
Are you running within a docker container?

technicsdj, 2024-11-11:
It currently looks as if this problem is caused by HA itself. There was an update to the Supervisor today, but this cannot be installed. I will wait and see when the update finally works. If it still doesn't work after that, I will contact you again. Until then, thanks for your help.

suaveolent, 2024-11-11:
Alright 👍

technicsdj, 2024-11-11:
My HA is running again so far. Unfortunately still the same error. I was able to install the addon for the PIP command, but, that didn't help either. It installs version 0.2.7, is that correct? 

`➜  ~ pip install hoymiles-wifi
Looking in indexes: https://pypi.org/simple, https://wheels.home-assistant.io/musllinux-index/
Collecting hoymiles-wifi
  Downloading hoymiles_wifi-0.2.7-py3-none-any.whl.metadata (4.4 kB)
Collecting protobuf (from hoymiles-wifi)
  Downloading https://wheels.home-assistant.io/musllinux-index/protobuf-5.28.3-cp312-cp312-musllinux_1_2_aarch64.whl (363 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 363.2/363.2 kB 7.4 MB/s eta 0:00:00
Collecting crcmod (from hoymiles-wifi)
  Downloading https://wheels.home-assistant.io/musllinux-index/crcmod-1.7-cp312-cp312-musllinux_1_2_aarch64.whl.metadata (5.2 kB)
Downloading hoymiles_wifi-0.2.7-py3-none-any.whl (33 kB)
Downloading https://wheels.home-assistant.io/musllinux-index/crcmod-1.7-cp312-cp312-musllinux_1_2_aarch64.whl (22 kB)
Installing collected packages: crcmod, protobuf, hoymiles-wifi`

suaveolent, 2024-11-11:
Yes that sounds about right. Is the pip command run in the same environment as your home assistant instance?

technicsdj, 2024-11-11:
I have executed the command via the Advanced SSH & Web Terminal addon directly in the HA interface

technicsdj, 2024-11-12:
I have now tried a little more and found the following.

I uninstalled hoymiles under hacs and deleted the configuration under integrations. this only worked after restarting ha.

then i added hoymiles again under hacs, this also worked. hoymiles must be added under the integrations. but this does not work and the following error message appears:

The configuration flow could not be loaded: {"message":"Invalid handler specified"}

suaveolent, 2024-11-12:
The message typically shows if the hoymiles_wifi lib cannot be installed. 
Is there anything in the logs?

technicsdj, 2024-11-12:
[image]

these are the only errors that ha reports.

are config files from your hoymiles plugin still available somewhere? i have read that other plugins with the same error still had config files. after deleting the config files, it could be installed.

suaveolent, 2024-11-12:
The only config file which is used is the `core.config_entries` under `.homeassistant/.storage.` You can check if there is any entry for hoymiles_wifi.

technicsdj, 2024-11-13:
unfortunately there is no entry to be found there.

is it possible to add it manually? do you have an example of the config entry here?

suaveolent, 2024-11-13:
Yes, I can provide you with a config but this will not solve the issue:

1.) The integration relies on the hoymiles_wifi lib. If it cannot be found (ìndicated with the `no module named` error) it won't work.
2.) You also need the serial number in the right format for it to work.

Best approach is to analyse why `hoymiles_wifi` lib cannot be found.

Can you try to issue the `pip install hoymiles-wifi` from the context of the home assistant installation? I would assume that it relies on some kind of venv. (I'm using home assistant core, so I cannot give you full guidance here).

suaveolent, 2024-11-13:
Another idea: Maybe the update broke some permissions or requires a newer version of hacs.

Have you tried upgrading/re-installing hacs?

suaveolent, 2024-11-13:
@technicsdj have you found this issue already?

https://github.com/home-assistant/core/issues/126859#issuecomment-2400679379

Could be worth a try, obviously replacing `roborock` with `hoymiles_wifi`

technicsdj, 2024-11-13:
I have now completely reinstalled hacs, but unfortunately that didn't help either.

I have already found the search command, but the system can't find anything there.

[image]

suaveolent, 2024-11-13:
Can you install other custom components which have dependencies to external libraries?

technicsdj, 2024-11-13:
I tried everything and nothing worked.

The only thing that helped was to make a backup of the system and reload homeassistant onto the SD card using Pi Imager. 

Now everything works again... thank you very much for your help :)

suaveolent, 2024-11-13:
Thanks for letting me know. It seems that the update messed up some permissions. 

Great that it works again 👍