suaveolent/hoymiles-wifi #37: Support for DTU W500 in MS-A2 storage

thomluther, 2024-11-12:
This is more a question than a feature request.
Hoymiles released an AC micro storage system MS-A2. According to videos I noticed that when the S-Miles App connects to it, it shows DTU W500
Since the system has Wifi integrated like the inverters (and internally there is also a Hoymiles Hybrid inverter for AC charge and discharge of the battery, there might be similar protocols used for the communication.
So the question is whether someone already tried to connect to it with your library and whether this MS system could be supported as well, mainly for value/parameter monitoring but also for management like changing mode and output power to realize custom built output regulation, e.g. via Home Assistant.

suaveolent, 2024-11-12:
That is a good question.

If there is an internal inverter one could try to issue the app-information-data and get-real-data-new request.

If that does not work I need to dig in the hoymiles app code or someone would have to sniff the packages.

krikk, 2024-12-27:
no luck with the hoymiles-wifi command only getting:

`No response or unable to retrieve response for app-information-data`

i did a nmap scan of my MS-A2 IP:
```
nmap -sS -p 1-65535 192.168.11.116
Starting Nmap 7.93 ( https://nmap.org ) at 2024-12-27 22:22 CET
Nmap scan report for 192.168.11.116
Host is up (0.012s latency).
Not shown: 65534 closed tcp ports (reset)
PORT      STATE SERVICE
43485/tcp open  unknown
MAC Address: xxxx (Unknown)

Nmap done: 1 IP address (1 host up) scanned in 52.89 seconds
```

when i connect to the given port with firefox i get some data which changes on every reload and always begins with:

`&D4…`

...i am not sure if this helps...

krikk, 2024-12-27:
i also did niff the app traffic from the iphone app, on my fritzbox, but as i see it, the Hoymiles MS-A2 connects via TLS to

`https://dataeu.hoymiles.com:10083/`

...and because of TLS, i do not see anything :(

DocSnyd3r, 2024-12-31:
I will order one myself, maybe it will noch check certificate and could be redirected...

krikk, 2025-01-01:
played aroud a bit with https://github.com/emanuele-f/PCAPdroid ...could sniff the traffic to the cloud of the android app, full logs will follow tomorrow:

```POST /pvmc/api/0/station_data/real_g_c HTTP/1.1
Authorization: xxxxxxxxxx
User-Agent: sma/ad/1.4.0/159/1
Charset: UTF-8
language: en_us
Content-Type: application/json; charset=utf-8
Content-Length: 17
Host: eud0.hoymiles.com
Connection: Keep-Alive
Accept-Encoding: gzip

{"sid":"xxxx"}

HTTP/1.1 200 OK
Date: Wed, 01 Jan 2025 20:59:14 GMT
Content-Type: application/json;charset=UTF-8
Content-Length: 386
Connection: keep-alive
Server: nginx
Content-Encoding: gzip
Vary: Accept-Encoding
X-Application-Context: pvm-consumer:prod:80
Access-Control-Allow-Headers: authorization,content-type,language
Access-Control-Allow-Methods: GET,POST,OPTIONS
Access-Control-Allow-Credentials: true
Content-Security-Policy: default-src 'self' 'unsafe-inline' 'unsafe-eval' https: blob: data:;
Strict-Transport-Security: max-age=31536000; includeSubDomains
via: EU-GER-frankfurt-EDGE5-CACHE6[707],EU-GER-frankfurt-EDGE5-CACHE6[ovl,691]
X-CCDN-REQ-ID-46B1: xxxxxxxxxxx

{"status":"0","message":"success","data":{"today_eq":"1","month_eq":"0","year_eq":null,"total_eq":"0.0","real_power":"0","co2_emission_reduction":"0","plant_tree":"0","data_time":"2025-01-01 21:52:30","last_data_time":"2025-01-01 21:52:30","capacitor":"2","is_balance":0,"is_reflux":1,"reflux_station_data":{"start_date":"2025-01-01","end_date":"2025-01-01","pv_power":"0","grid_power":"0","load_power":"0","bms_power":"0","bms_soc":"60.5","inv_num":1,"meter_location":2,"pv_to_load_eq":"0","load_from_pv_eq":"0","meter_b_in_eq":"0","meter_b_out_eq":"0","bms_in_eq":"0","bms_out_eq":"0","self_eq":"0","pv_eq_total":"0","use_eq_total":"0","flows":[],"icon_pv":0,"icon_grid":1,"icon_load":1,"icon_bms":1,"icon_gen":0,"icon_pvi":0,"mb_in_eq":{"today_eq":"0","month_eq":"0","year_eq":"0","total_eq":"8"},"mb_out_eq":{"today_eq":"0","month_eq":"0","year_eq":"0","total_eq":"0"},"cfg_load_power":0,"icon_plug":0,"icon_ai_plug":0}}}```

"bms_soc":"60.5" <-- this is my battery SOC :)

DocSnyd3r, 2025-01-02:
ganz interessant Richtung Clou geht wohl
`{
  "h": 1902814347,
  "uris": [
    {
      "key": "PVMC_STATION_DATA_REAL_G_C",
      "uri": "/pvmc/api/0/station_data/real_g_c",
      "action": 1,
      "did": 10
    },
    {
      "key": "PVMC_STATION_DATA_EQ_MONTH_YEAR_S_C",
      "uri": "/pvmc/api/0/station_data/eq_month_year_s_c",
      "action": 1,
      "did": 10
    },
    {
      "key": "PVMC_STATION_DATA_EQ_TOTAL_YEAR_S_C",
      "uri": "/pvmc/api/0/station_data/eq_total_year_s_c",
      "action": 1,
      "did": 10
    },
    {
      "key": "PVMC_STATION_DATA_EQ_DAY_MONTH_S_C",
      "uri": "/pvmc/api/0/station_data/eq_day_month_s_c",
      "action": 1,
      "did": 10
    },
    {
      "key": "PVMC_STATION_DATA_P_DAY_S_C",
      "uri": "/pvmc/api/0/station_data/p_day_s_c",
      "action": 1,
      "did": 10
    },
    {
      "key": "FW_UPG_PATCH_LATEST_S_C",
      "uri": "/fw/pub/0/dev/upg_patch/latest_s_c",
      "action": 1,
      "did": 0
    },
    {
      "key": "ADVERT_ADVERT_LATEST_S_A",
      "uri": "/advert/pub/0/advert/latest_s_a",
      "action": 1,
      "did": 0
    },
    {
      "key": "MONITOR_BAN_WBMW",
      "uri": "/monitor/api/0/ban/wbmw",
      "action": 1,
      "did": 0
    },
    {
      "key": "MONITOR_BAN_MIB",
      "uri": "/monitor/api/0/ban/mib",
      "action": 1,
      "did": 0
    },
    {
      "key": "MONITOR_BAN_GMWB",
      "uri": "/monitor/api/0/ban/gmwb",
      "action": 1,
      "did": 0
    },
    {
      "key": "PVMC_DTU_B_U_C",
      "uri": "/pvmc/api/0/dtu/b_u_c",
      "action": 1,
      "did": 0
    },
[... 501 lines cut ...]
      "action": 1,
      "did": 0
    },
    {
      "key": "PVMC_MICRO_DELETE_C",
      "uri": "/pvmc/api/0/micro/delete_c",
      "action": 1,
      "did": 0
    }
  ],
  "dos": [
    {
      "domain": "https://euapi.hoymiles.com",
      "id": 1
    },
    {
      "domain": "https://neapi.hoymiles.com",
      "id": 0
    },
    {
      "domain": "https://eud0.hoymiles.com",
      "id": 10
    }
  ]
}`

krikk, 2025-01-02:
i used an android studio emulator with android 12... 

then followed basically this guide: https://emanuele-f.github.io/PCAPdroid/tls_decryption#34-caveats-and-possible-solutions 

1. add adb to path, to make the rootAVD script work
2. ...used https://gitlab.com/newbit/rootAVD to root the emulator and install magisk
3. in magisk i installed this module: https://github.com/NVISOsecurity/MagiskTrustUserCerts <- this adds user imported certs to system trust store... 
4. install PCAPdroid and install CA Certificate
5. rebootet emulator (the magisk module only copies user the certificate to system store on boot)

and afterwards TLS decryption with PCAPdroid works...

i speak german, but stay with english here... :)

krikk, 2025-01-02:
sniffed the login process:

```
POST /iam/pub/0/c/login_c HTTP/1.1
User-Agent: sma/ad/1.4.0/159/0
Charset: UTF-8
language: en_us
Content-Type: application/json; charset=utf-8
Content-Length: 123
Host: euapi.hoymiles.com
Connection: Keep-Alive
Accept-Encoding: gzip

{"user_name":"some-email@gmail.com","password":"b8e134c5cb0dff96842724b7cb238500.FEMBBJ32xHaBEMWJk81VLFJs7+Oc2RIhq\/4did7zVwA="}
HTTP/1.1 200 OK
Date: Thu, 02 Jan 2025 19:33:46 GMT
Content-Type: application/json;charset=UTF-8
Content-Length: 112
Connection: keep-alive
Server: nginx/1.22.1
Vary: Accept-Encoding
Access-Control-Allow-Headers: authorization,content-type,language
Access-Control-Allow-Methods: GET,POST,OPTIONS
Access-Control-Allow-Credentials: true
via: EU-ITA-milan-EDGE2-CACHE8[16],EU-ITA-milan-EDGE2-CACHE8[ovl,14]
X-CCDN-REQ-ID-46B1: xxxx

{"status":"0","message":"success","data":{"token":"2.xxxxxxxxxxxxxxxxxx.1"}}
POST /dict/pub/0/app_ver/update_check HTTP/1.1
Authorization: 2.xxxxxxxxxxxxxxxxxx.1
User-Agent: sma/ad/1.4.0/159/1
Charset: UTF-8
language: en_us
Content-Type: application/json; charset=utf-8
Content-Length: 30
Host: euapi.hoymiles.com
Connection: Keep-Alive
Accept-Encoding: gzip

{"type":2,"name":"AURORA_APP"}
```
the password in the example above is "NoCloud4Hoymiles"

the password is encoded like this:
{md5-hash-of-password}.{sha-256-hash of password with output encoding base64}

used this site to find hashes: https://emn178.github.io/online-tools/sha256.html

bash command to generate password string: 
```
echo -n "$(echo -n "NoCloud4Hoymiles" | md5sum | awk '{print $1}').$(echo -n "NoCloud4Hoymiles" | openssl dgst -sha256 -binary | openssl base64)"
```

curl command to do login, this will get you a authorization token which you will need in the next commands:
```
curl -H 'Content-Type: application/json; charset=utf-8' -d '{"user_name":"some-email@gmail.com","password":"b8e134c5cb0dff96842724b7cb238500.FEMBBJ32xHaBEMWJk81VLFJs7+Oc2RIhq/4did7zVwA="}' -X POST "https://euapi.hoymiles.com/iam/pub/0/c/login_c"
```
output:
```
{"status":"0","message":"success","data":{"token":"2.xxxxxxxxxxxxxxx.1"}}
```

get the sid, fill in the token in the authorization header, first
```
curl -H 'Content-Type: application/json; charset=utf-8' -H 'Authorization: 2.xxxxxxx.1' -d '{"page":1,"page_size":50}' -X POST "https://neapi.hoymiles.com/pvmc/api/0/station/select_by_page_c"
```
output:
```
{"status":"0","message":"success","data":{"page":1,"page_size":50,"total":1,"list":[{"sid":xxxxx,"sn":"serial","name":"tester-home","area_code":"AT","classify":10,"devices":[{"id":75460,"sn":"serial","dtu_sn":"serial","type":1,"warn_data":{"connect":true,"warn":false},"extend_data":{"dfs":19211107,"inner":1},"devices":[{"id":xxx,"sn":"serial","dtu_sn":"serial","type":6,"warn_data":{"connect":true,"warn":false},"extend_data":{"role":0},"devices":[]}]}],"dc":1,"ak":"xxxx","bt":1}]}}
```

with the sid get the uri?
```
curl -H 'Content-Type: application/json; charset=utf-8' -H 'Authorization: 2.xxxxxxxxx.1' -d '{"sid":FILLYOURSID}' -X POST "https://neapi.hoymiles.com/pvmc/api/0/station/get_sd_uri_c"
```
output gives the uri for the next request:
```
{"status":"0","message":"success","data":{"uri":"https://eurt.hoymiles.com/rds/api/0/burst/get?k=1658h352ExxxxxxxV8Moz&t=173xxxxx4"}}
```

with the uri we finally can get the data we want, SOC of the battery:
```
curl -H 'Content-Type: application/json; charset=utf-8' -H 'Authorization: 2.xxxxxxxxx.1' -d '{"m":0,"sid":7500620}' -X POST "https://eurt.hoymiles.com/rds/api/0/burst/get?k=1658h352Ebn0aa8JFhVdqywnIlDZ937m3gQj07KmlR6rW1l2MPJkRyDR0iSrsndQWPcqA6co9yMCbaK9pnxuDcI6qq5vp9m0Rvg7yhY&t=1735851466351"
```
output:
```
{"status":"0","data":{"flow":[],"dly":1500,"con":1,"soc":60.5,"power":{"pv":0.0,"bat":0.0,"grid":0.0,"load":0.0,"sp":0.0},"brs":0,"bhs":0,"ems":0}}
```

@DocSnyd3r: can you reproduce this??

krikk, 2025-01-02:
get/set on-grid input settings:
```
curl -H 'Content-Type: application/json; charset=utf-8' -H 'Authorization: 2.xxxxxxx.1' -d '{"action":1513,"data":{"sid":FILLSID}}' -X POST "https://neapi.hoymiles.com/pvmc/api/0/station_ctl/read_c"

curl -H 'Content-Type: application/json; charset=utf-8' -H 'Authorization: 2.xxxxxx.1' -d '{"id":"21031163"}' -X POST "https://neapi.hoymiles.com/pvmc/api/0/station_ctl/setting_status_c"

curl -H 'Content-Type: application/json; charset=utf-8' -H 'Authorization: 2.xxxxx.1' -d '{"action":1513,"data":{"power":388,"sid":7500620}}' -X POST "https://neapi.hoymiles.com/pvmc/api/0/station_ctl/write_c"

```

JD92-bit, 2025-01-02:
I can reproduce every command besides
`curl -H 'Content-Type: application/json; charset=utf-8' -H 'Authorization: 2.xxxxxx.1' -d '{"id":"21031163"}' -X POST "https://neapi.hoymiles.com/pvmc/api/0/station_ctl/setting_status_c"`
where I get the following message 

> {"status":"1","message":"system error","data":null}

But that could be because I don't have the device plugged into the grid.

suaveolent, 2025-01-03:
Wow thank you for all your hard work. I am wondering, since you are not communicating directly with the W500 but rather the cloud API, whether this is similar to what is done here?

https://github.com/dmslabsbr/hoymiles

Also I would be interested in whether there is some kind of "toolkit" in the according app which would allow you to see live data locally?

I most likely will decompile the android app next week to see if I can find anything useful.

krikk, 2025-01-03:
thanks for the link to https://github.com/dmslabsbr/hoymiles ...you are right, that repo seems to be more related, do my findings... perhaps i will open an "issue" over there...

my hope at start was also to communicate directly with the MS-A2 in LAN, but as stated above, the MS-A2 only has one port open which gives us "data" , starting with "&D4" ...but rest not readable for me...

so my next step was to see what the app does, and as fare as i have seen, the "S-Miles Home" App on Android only "talks" to the hoymiles-Cloud, have not seen any local connections (in normal app usage)

...the first "pairing" of the App with the factory-default MS-A2 seems to happen via Bluetooth, did not have a look on this...

...will also try to sniff traffic when integrating my Shelly Pro 3EM (must install it first)

DocSnyd3r, 2025-01-03:
API is working for me also. The other repo seems to use different password hasing seems md5 hex value representation which is not or no longer working.
For the local port I get the 168 bytes back where the first 8 are always "2644348500000090" the rest seems completely random. Later today my Shelly em3 pro arrives but I guess this will only be used to pull data. Since the data after the first 8 bytes seems completely random a firmware to decompile would be nice.

DocSnyd3r, 2025-01-12:
So in a different forum a user flagged as vendor sayd next feature is AI followed by MQTT, whatever this is worth.

krikk, 2025-01-12:
with the help of chatGPT i was able to produce this python3 script which publishes the SOC  and a few other data to an MQTT server: (not pretty, but working)

need to install python mqtt library first...
`sudo apt install python3-paho-mqtt`

the python3 code:

```python
import hashlib
import base64
import requests
import paho.mqtt.client as mqtt

# Configuration variables
hoymiles_user = "user"
hoymiles_password = "SECRET"
mqtt_broker = "192.168.1.23"
mqtt_user = "mqtt-user"
mqtt_password = "mqtt-password"
mqtt_topic = "hoymiles-ms-a2"

# Step 1: Generate the MD5 hash
md5_hash = hashlib.md5(hoymiles_password.encode('utf-8')).hexdigest()

# Step 2: Generate the SHA-256 hash and encode it in Base64
sha256_hash = hashlib.sha256(hoymiles_password.encode('utf-8')).digest()
base64_hash = base64.b64encode(sha256_hash).decode('utf-8')

# Combine the two parts to form the final string
final_string = f"{md5_hash}.{base64_hash}"

# Step 3: Send the first web request with the generated string
url_login = "https://euapi.hoymiles.com/iam/pub/0/c/login_c"
headers = {'Content-Type': 'application/json; charset=utf-8'}

data_login = {
    "user_name": hoymiles_user,
    "password": final_string
}

# Send the POST request for login
response_login = requests.post(url_login, json=data_login, headers=headers)

# Check if the response is successful
if response_login.status_code == 200:
    response_data = response_login.json()
    if response_data.get("status") == "0" and "data" in response_data:
        token = response_data["data"].get("token")
        if token:
            print("Token retrieved:", token)

            # Step 4: Use the token to send the second request
            url_station = "https://neapi.hoymiles.com/pvmc/api/0/station/select_by_page_c"
            headers_with_auth = {
                'Content-Type': 'application/json; charset=utf-8',
                'Authorization': token
            }

            data_station = {
                "page": 1,
                "page_size": 50
            }

            # Send the POST request for station data
            response_station = requests.post(url_station, json=data_station, headers=headers_with_auth)

            # Check if the station response is successful
            if response_station.status_code == 200:
                station_data = response_station.json()
                print("Station Data Response:", station_data)
[... 62 lines cut ...]
                                            print("Failed to retrieve final data:", final_data_response.get("message"))
                                    else:
                                        print("Failed to fetch final data. Status Code:", response_final.status_code)
                                else:
                                    print("URI not found in sd_uri response.")
                            else:
                                print("Failed to retrieve sd_uri data:", sd_uri_data.get("message"))
                        else:
                            print("Failed to fetch sd_uri. Status Code:", response_sd_uri.status_code)
                    else:
                        print("SID not found in station data response.")
                else:
                    print("Failed to retrieve station data:", station_data.get("message"))
            else:
                print("Failed to fetch station data. Status Code:", response_station.status_code)
        else:
            print("Token not found in response.")
    else:
        print("Login failed:", response_data.get("message"))
else:
    print("Failed to login. Status Code:", response_login.status_code)

```

feel free to give it a try

DocSnyd3r, 2025-01-13:
AI fell in love with all the if statements :)
I'd say one would reuse the token until it expires and only then get a new token, the IdP server may not like the constant token polling.

krikk, 2025-01-13:
thought about the token "re-use" already, but ran out of free credits with chatGPT...

...lets hope hoymiles really gives us a firmware with MQTT support, i really would prefer to query all this data locally...

krikk, 2025-01-22:
create a github repo with a "basic" script, to get cloud data to mqtt: https://github.com/krikk/hoymiles-ms-a2-to-mqtt

suaveolent, 2025-02-26:
@krikk I have been working on supporting Battery based inverters from hoymiles. 

Can you check whether the new requests also work for the DTU W500?

https://github.com/suaveolent/ha-hoymiles-wifi/issues/88#issuecomment-2665208639

Specifically I would be interested in the following commands:

`get-gateway-info`
`get-gateway-network-info` 
`get-energy-storage-registry`

Thanks!

krikk, 2025-02-26:
i tried it, and the new build works, because i can get data from my HMS-800W-2T... 

but i CAN NOT get any data from my Hoymiles MS A2, and this once again confirms to me that the "Hoymiles MS A2" does NOT have a DTU like the other products :(

my nmap scan of my HMS-800W-2T:
```
[22:09:21] # nmap -sS -p 1-65535 192.168.11.73
Starting Nmap 7.93 ( https://nmap.org ) at 2025-02-26 08:42 CET
Nmap scan report for 192.168.x
Host is up (0.0089s latency).
Not shown: 65534 closed tcp ports (reset)
PORT      STATE SERVICE
10081/tcp open  famdc
MAC Address: 54:F2:9F:xxx (Unknown)
Nmap done: 1 IP address (1 host up) scanned in 66.49 seconds
```

my Hoymiles MS A2:
```
[08:44:00] # nmap -sS -p 1-65535 192.168.11.116
Starting Nmap 7.93 ( https://nmap.org ) at 2025-02-26 08:44 CET
Nmap scan report for 192.168.x
Host is up (0.013s latency).
Not shown: 65534 closed tcp ports (reset)
PORT      STATE SERVICE
43485/tcp open  unknown
MAC Address: 44:87:63:xxx (Unknown)
Nmap done: 1 IP address (1 host up) scanned in 49.40 seconds
```

suaveolent, 2025-02-26:
Thanks for confirming! I will close this issue in that case!

suaveolent, 2025-02-26:
@krikk what I noticed, though is the open port `43485` which I suspect could be a Modbus TCP Port (Note the 485)

closed 2025-02-26 by suaveolent

suaveolent, 2025-02-26:
@krikk maybe https://github.com/[wasilukm/hoymiles_modbus](https://github.com/wasilukm/hoymiles_modbus) is what you are looking for