suaveolent/hoymiles-wifi #39: Only 2 out of 3 microinverters being listed in get-real-data-new

sergiou87, 2024-11-19:
**Describe the bug**

When I run `hoymiles-wifi --host 192.168.0.19 identify-inverters` I only get 2 out of the 3 microinverters I have installed. I looked at the implementation and saw it comes from the `sgs_data` entries of `get-real-data-new`, so I ran `hoymiles-wifi --host 192.168.0.19 get-real-data-new` and I only get 2 entries of `sgs_data` with the data of 2 microinverters.

As mentioned, I have 3 microinverters installed and the `S-Miles Enduser` app is able to see the 3 of them, I have no idea what's going on or how can I debug this 😕 

I waited for a few hours just in case it showed up later, but no luck.

**To Reproduce**
Steps to reproduce the behavior:
1. Run `hoymiles-wifi --host 192.168.0.19 identify-inverters`

**Expected behavior**
3 microinverters

**Screenshots**

Output of `identify-inverters` (masking the serial numbers in case they're sensitive info in some way 😅 ):

```
Identify-inverters Response:
{'1164a00*****': 'HMS-1600/1800/2000-4T', '1164a00*****': 'HMS-1600/1800/2000-4T'}
```

Output of `get-real-data-new`:

```
Get-real-data-new Response: 
device_serial_number: "**********"
timestamp: 1732012667
active_power: 4
firmware_version: 1
meter_data {
  device_type: 1
  serial_number: **********
  phase_total_power: 15
  phase_A_power: 15
  power_factor_total: 305
  energy_total_power: 149
  energy_phase_A: 149
  energy_total_consumed: 916
  energy_phase_A_consumed: 916
  fault_code: 1
  voltage_phase_A: 22830
  current_phase_A: 250
  power_factor_phase_A: 305
}
sgs_data {
  serial_number: **********
  firmware_version: 1
  voltage: 2276
  frequency: 4998
  active_power: 3592
  current: 157
  power_factor: 1000
  temperature: 186
  warning_number: 1
  crc_checksum: 28864
  link_status: 1
  modulation_index_signal: -83
}
sgs_data {
  serial_number: **********
  firmware_version: 1
  voltage: 2286
  frequency: 4999
  active_power: 3710
  current: 162
  power_factor: 1000
  temperature: 187
  warning_number: 1
  link_status: 1
  modulation_index_signal: -83
}
pv_data {
  serial_number: **********
  port_number: 1
  voltage: 406
  current: 231
  power: 942
  energy_total: 318
  energy_daily: 132
  error_code: 50921472
}
pv_data {
  serial_number: **********
  port_number: 2
  voltage: 407
  current: 231
  power: 942
  energy_total: 321
  energy_daily: 132
  error_code: 50331648
}
pv_data {
  serial_number: **********
  port_number: 3
  voltage: 408
  current: 232
  power: 950
  energy_total: 324
  energy_daily: 133
  error_code: 50331648
}
pv_data {
  serial_number: **********
  port_number: 4
  voltage: 408
  current: 232
  power: 946
  energy_total: 322
  energy_daily: 133
  error_code: 50331648
}
pv_data {
  serial_number: **********
  port_number: 1
  voltage: 409
  current: 239
  power: 981
  energy_total: 329
  energy_daily: 134
  error_code: 50921472
}
pv_data {
  serial_number: **********
  port_number: 2
  voltage: 408
  current: 239
  power: 977
  energy_total: 333
  energy_daily: 133
  error_code: 50331648
}
pv_data {
  serial_number: **********
  port_number: 3
  voltage: 406
  current: 238
  power: 970
  energy_total: 324
  energy_daily: 132
  error_code: 50331648
}
pv_data {
  serial_number: **********
  port_number: 4
  voltage: 407
  current: 239
  power: 976
  energy_total: 331
  energy_daily: 133
  error_code: 50331648
}
dtu_power: 10868
dtu_daily_energy: 1590
```

List of microinverters from S-Miles Enduser app

[image]

sergiou87, 2024-11-19:
I have some software engineering/debugging skills, so let me know if there is anything I can do to help 🙏

suaveolent, 2024-11-19:
Thanks, you could actually try the `multi-inverter` branch. I pushed a fix for this yesterday, but cannot test it since I only have a single inverter.

sergiou87, 2024-11-19:
Wow! Perfect timing then 😄 That branch works like a charm 🙇‍♂ 

```
$ hoymiles-wifi --host 192.168.0.19 --as-json identify-inverters
{
    "1164a00*****": "HMS-1600/1800/2000-4T",
    "1164a00*****": "HMS-1600/1800/2000-4T",
    "1164a00*****": "HMS-1600/1800/2000-4T"
}
```

suaveolent, 2024-11-19:
Great news. I will soon prepare a new release then.

suaveolent, 2024-11-19:
I have just released version `v0.3.0` which fixes this issue.

closed 2024-11-19 by suaveolent

sergiou87, 2024-11-19:
Nice! Thank you, looking forward to 0.3.3 of ha-hoymiles-wifi too 🙏🙏🙏