suaveolent/hoymiles-wifi #41: Wrong or missing consumption data? 🤔

sergiou87, 2024-11-20:
Not sure if I'm missing something here, but the info about my meter provided by the S-Miles Enduser app doesn't match the output of `hoymiles-wifi`:

App vs ha-hoymiles-wifi:
[image]

hoymiles-wifi output:

```
$ hoymiles-wifi --host 192.168.0.19 get-real-data-new --as-json
{
  "deviceSerialNumber": "******",
  "timestamp": 1732109181,
  "activePower": 4,
  "firmwareVersion": 1,
  "meterData": [
    {
      "deviceType": 1,
      "serialNumber": "******",
      "phaseTotalPower": 261,
      "phaseAPower": 261,
      "powerFactorTotal": 983,
      "energyTotalPower": 1959,
      "energyPhaseA": 1959,
      "energyTotalConsumed": 1972,
      "energyPhaseAConsumed": 1972,
      "faultCode": 1,
      "voltagePhaseA": 23170,
      "currentPhaseA": 1149,
      "powerFactorPhaseA": 983
    }
  ],
...
```

suaveolent, 2024-11-20:
Can you update to `main` branch? There are some conversion erros in the `0.3.2` release, since I don't have a meter and had to guess the units. `main` branch (or next release) should adresses it.

sergiou87, 2024-11-20:
If I had to guess, I'd say things got worse:

[image] [image]

```json
  "meterData": [
    {
      "deviceType": 1,
      "serialNumber": "*****",
      "phaseTotalPower": 72,
      "phaseAPower": 72,
      "powerFactorTotal": 846,
      "energyTotalPower": 2200,
      "energyPhaseA": 2200,
      "energyTotalConsumed": 1972,
      "energyPhaseAConsumed": 1972,
      "faultCode": 1,
      "voltagePhaseA": 22980,
      "currentPhaseA": 385,
      "powerFactorPhaseA": 846
    }
  ],
```

suaveolent, 2024-11-21:
Huh. That's weird. Is there anything in the logs? Seems like the data cannot be read any more from the meter.

suaveolent, 2024-11-22:
I thinks the consumption is calculated via inverter power and meter power. No direct value from meter.

sergiou87, 2024-12-02:
I'm back 😅 Now everything looks good, so maybe it was an exceptional issue, but I think using the `main` branch was key for this.

I think I understand now how to get the consumption via the total generated power (by my 3 inverters) - the meter power (divided by the meter power factor?) so I created a small dashboard with all my panels and the rest of the info provided by the S-Miles app (probably need to fix a few issues though):

[image]

Generated power, consumed, injected to the grid, bought from the grid, then the solar panels, and then some graphs (with some crazy numbers as you can see, still don't know why).

Looking forward to the next version of `ha-hoymiles-wifi`! Thank you again 🙇‍♂

closed 2024-12-02 by sergiou87