suaveolent/hoymiles-wifi #42: Lack of documentation

Toug19, 2024-11-20:
I opened this "Feature request" because I don't know where to post questions I have on using this python module...
I have 8 panels with 2 micro inverters HMS-1600-4T (4 panels each).
These inverters are not wifi capable (from what I know, but not even sure). They are connected to a DTU-Lite-S which is connected to my home wifi.
From what I understood, this python package should be used only with inverters that directly use wifi. But I am able to retrieve data with it. From at least 6 commands I tested there is only 1 command which return error: app-get-hist-power.
I don't think this is a big deal.

The thing is I don't understand the returned values from the other commands. There are no units.
I want to get the current produced power.
I think the command "get-real-data-new" is the best to use. But here is a sample of data returned when production is very low (around 300W):

pi@raspberrypi:~ $  :hoymiles-wifi --host "192.168.1.155" get-real-data-new
Get-real-data-new Response: 
device_serial_number: "4130A38100B6"
timestamp: 1732111972
active_power: 1
firmware_version: 1
sgs_data {
  serial_number: 19123696322610
  firmware_version: 1
  voltage: 2375
  frequency: 5000
  active_power: 1317
  reactive_power: 1886
  current: 55
  power_factor: 572
  temperature: 157
  warning_number: 4
  crc_checksum: 28864
  link_status: 1
  modulation_index_signal: 100
}
sgs_data {
  serial_number: 19123696322352
  firmware_version: 1
  voltage: 2373
  frequency: 5000
  active_power: 1337
  reactive_power: 1880
  current: 56
  power_factor: 579
  temperature: 157
  warning_number: 3
  crc_checksum: 28864
  link_status: 1
  modulation_index_signal: 100
}
pv_data {
  serial_number: 19123696322610
  port_number: 1
  voltage: 331
  current: 102
  power: 338
  energy_total: 7352
  energy_daily: 180
  error_code: 50921472
}
pv_data {
  serial_number: 19123696322610
  port_number: 2
  voltage: 336
  current: 105
  power: 354
  energy_total: 7504
  energy_daily: 189
  error_code: 50331648
}
pv_data {
  serial_number: 19123696322610
  port_number: 3
  voltage: 334
  current: 105
  power: 352
  energy_total: 7439
  energy_daily: 189
  error_code: 50331648
}
pv_data {
  serial_number: 19123696322610
  port_number: 4
  voltage: 332
  current: 102
  power: 340
  energy_total: 7305
  energy_daily: 182
  error_code: 50331648
}
pv_data {
  serial_number: 19123696322352
  port_number: 1
  voltage: 332
  current: 103
  power: 344
  energy_total: 7282
  energy_daily: 184
  error_code: 50921472
}
pv_data {
  serial_number: 19123696322352
  port_number: 2
  voltage: 337
  current: 105
  power: 356
  energy_total: 7436
  energy_daily: 190
  error_code: 50331648
}
pv_data {
  serial_number: 19123696322352
  port_number: 3
  voltage: 336
  current: 104
  power: 352
  energy_total: 7401
  energy_daily: 188
  error_code: 50331648
}
pv_data {
  serial_number: 19123696322352
  port_number: 4
  voltage: 335
  current: 105
  power: 355
  energy_total: 7418
  energy_daily: 190
  error_code: 50331648
}

This values changes around the time, but can understand them without documentation.
pv_data? panel data?

suaveolent, 2024-11-20:
Thanks, that is a good and valuable point. I have not yet come around on documenting the values. But this is definately something which needs to be adressed.

In the meantime you can have a look at how the Home Assistant integration handles these values as well as their units (starting from line 114):

https://github.com/suaveolent/ha-hoymiles-wifi/blob/ef2180cd7c5f2348a57d09610c6de220f2be7788/custom_components/hoymiles_wifi/sensor.py#L114

suaveolent, 2024-11-20:
Have a look at the main.py how to invoke from asynchronous context.

Toug19, 2024-11-21:
Thanks. I have all the info I need for the dev I plan.

closed 2024-11-21 by Toug19