suaveolent/hoymiles-wifi #44: Requests fail randomly

sergiou87, 2024-12-08:
I have 3 microinverters and I noticed that one of them was intermittently unavailable, always the same one. After some debugging, I realized that microinverter was randomly missing from the `async_get_real_data_new` response. After more digging, I saw that for me `async_get_real_data_new` requires 4 requests and it sometimes failed with the request with that microinverter.

I've seen it fail because of:
- [Buffer is too short for unpacking](https://github.com/suaveolent/hoymiles-wifi/blob/a495cc6660dcb9cccd7c985324a5ce14c1265508/hoymiles_wifi/dtu.py#L481)
- [OSError / asyncio.TimeoutError](https://github.com/suaveolent/hoymiles-wifi/blob/a495cc6660dcb9cccd7c985324a5ce14c1265508/hoymiles_wifi/dtu.py#L464) (I don't know which one exactly, though)

I've tried adding a simple retry strategy based on exponential backoff and so far so good, but sometimes it took up to 4 retries to get the response 😐 

Is this a known issue? Does the S-Miles app do anything like this to handle failed requests?

```diff
@@ -377,23 +389,46 @@ class DTU:
             command, request, CommandPB_pb2.CommandReqDTO
         )
 
     async def async_send_request(
         self,
         command: bytes,
         request: Any,
         response_type: Any,
         dtu_port: int = DTU_PORT,
     ):
-        """Send request to DTU."""
+        """Send request to DTU (with exponential backoff retries)."""
+
+        retries = 0
+        while retries < 5:
+            response = await self.async_send_single_request(
+                command, request, response_type, dtu_port
+            )
+            if response is not None:
+                return response
+            retries += 1
+            await asyncio.sleep(2 ** retries)
+
+        return None
+
+    async def async_send_single_request(
+        self,
+        command: bytes,
+        request: Any,
+        response_type: Any,
+        dtu_port: int = DTU_PORT,
+    ):
+        """Send single request to DTU (no retries, use async_send_request instead)."""
 
         self.sequence = (self.sequence + 1) & 0xFFFF
```

suaveolent, 2024-12-09:
It might actually be the other way around: we might need to introduce a delay inbetween the get-real-data-new requests as not to overwhelm the DTU.

sergiou87, 2024-12-09:
But even with a delay between requests, they could fail, so it'd make sense to add some retry strategy, right?. Does the mobile app do anything about this? (reading other messages of yours, I assume you reverse-engineered it)

suaveolent, 2024-12-10:
As far as I know there is no such mechanism in the App. However, I'll check again.

suaveolent, 2024-12-10:
@sergiou87 Correction, Indeed there is a delay in between the function calls of 2 seconds.

I will implement this feature and push a new version.

sergiou87, 2024-12-10:
Oh! That's great, thank you! I've been using [these changes](https://github.com/suaveolent/hoymiles-wifi/commit/f820a18768e5ca741f6e625957f5e0f8d1763806) (1 second delay between requests + exponential backoff for up to 5 retries) and so far so good, but I will try your change tomorrow!

Oh and if you want some extra eyes on how that app works feel free to share the source with me (or not!), my email address is [image]

suaveolent, 2024-12-11:
> Oh! That's great, thank you! I've been using [these changes](https://github.com/suaveolent/hoymiles-wifi/commit/f820a18768e5ca741f6e625957f5e0f8d1763806) (1 second delay between requests + [...]

Sure thing. However, the decompiled Android app is rather large (around 0.5-1Gb). The easiest way to get access is to decompile it yourself. It's not complicated:

1. Get APKLab:  https://github.com/APKLab/APKLab
2. Get the APK from e.g. APKPure: https://apkpure.com/de/s-miles-installer/com.hm.hemaiInstall1
3. Decompile :) 

The structure of the App is rather opaque. Best is to search for some known function calls (e.g. `setCp(`)  to figure out what is going on. If you need any more help just let me know.

sergiou87, 2024-12-11:
Thanks @suaveolent  🙇‍♂  Unfortunately I'm not getting anything useful… after decompiling it, it only takes about 70MB (far from those 0.5-1GB you mention). I've looked for `setCp(`, for example, no luck either. I've used these options:
[image]
 But also tried many other combinations with no luck. Any idea what could be wrong here?

suaveolent, 2024-12-11:
It seems like there is an issue with the `V1.1.32`. Maybe the package uploaded to APKPure is corrupted.
 Try with `V1.1.31`, this just worked for me.

sergiou87, 2024-12-11:
Yes, that one works!! Thank you 😄 

I've tested ha-hoymiles-wifi with my changes (2s delay between requests + exponential backoff retries) for almost 12h and I didn't see a single retry, so maybe those 2s delay are more than enough? 👏 

I'm gonna close this, and will try v0.3.2 directly with your changes, just to be sure it works. Otherwise I will let you know.

Thanks again!

closed 2024-12-11 by sergiou87

suaveolent, 2024-12-11:
That sounds great :) 
So 2s in between requests might be the absolute minimum.

suaveolent, 2024-12-19:
Thanks. 

Does it affect only subsequent calls or could also the initial request? 
I am wondering wether these errors might be due to the remaining calls overlapping with the get-real-data-new requests.

sergiou87, 2024-12-20:
Sometimes it also affects the initial request 😕 

Just to clarify, I haven't had exponential backoff retries during the last ~10 days, I'm using your latest changes + some logging stuff.

I guess it can also overlap with requests made from the phone app, right?

suaveolent, 2024-12-20:
Thanks, that’s what I though. I need to implement the 2s delay on request basis on not only for the get-real-data-new.

suaveolent, 2024-12-20:
@sergiou87 I have pushed a potential fix to `dev` branch. Can you maybe check that out?

sergiou87, 2024-12-20:
Done! I'll leave it running for a few days too 👍  However those changes don't prevent requests from failing when they overlap with the iOS/Android app, right? Hence why I think a retry strategy would be nice too.

I think that with all these recent changes, the retry strategy will be barely used, and only in those cases where strictly necessary.

suaveolent, 2024-12-20:
Yes that’s true. Once that is confirmed I will add a retry strategy too.

sergiou87, 2024-12-25:
Status update: it has worked flawlessly for days but last night a request failed. It was the initial request what failed. Hope that helps!