suaveolent/hoymiles-wifi #48: Is there a table of  warning codes?

kumy, 2024-12-30:
**Is your feature request related to a problem? Please describe.**
Based on your work I've built a simple "echo server" so my inverters (which are not allowed to go on internet) think they are connected to internet. Looks like doing so prevent them from disconnecting/reconnecting from the wifi at very regular intervals...

I've noticed they sometime send messages of type `WarnReq` and I'm curious to understand what are they reporting as warning.

Here are some examples:
```
2024-12-30 08:10:16.385342 | DTU -> CLOUD | command 220B (WarnReq) | seq 000F | {
  "dtu_sn": "4143A0xxxxxx",
  "time": 901,
  "package_nub": 1,
  "warn_device": 1,
  "warns": [
    {
      "pv_sn": "220702xxxxxxxx",
      "code": 32769,    <-----
      "num": 1,
      "s_time": -126,
      "e_time": -126
    },
    {
      "pv_sn": "220702xxxxxxxx",
      "code": 16514,    <-----
      "e_time": -1
    }
  ]
}

2024-12-29 17:00:01.604408 | DTU -> CLOUD | command 220B (WarnReq) | seq 027C | {
  "dtu_sn": "4143A0xxxxxx",
  "time": 1735488002,
  "package_nub": 1,
  "warn_device": 1,
  "warns": [
    {
      "pv_sn": "220702xxxxxxxx",
      "code": 45058,       <-----
      "num": 2,
      "s_time": 1735515785,
      "e_time": 1735515785,
      "w_data1": 65535,    <-----
      "w_data2": 61936     <-----
    },
    {
      "pv_sn": "220702xxxxxxxx",
      "code": 45058,
      "num": 3,
      "s_time": 1735515793,
      "e_time": 1735515793,
      "w_data2": 3601
    },
    {
      "pv_sn": "220702xxxxxxxx",
      "code": 45058,
      "num": 4,
      "s_time": 1735515845,
      "e_time": 1735515845,
      "w_data1": 65535,
      "w_data2": 61935
    },
    {
      "pv_sn": "220702xxxxxxxx",
      "code": 45058,
      "num": 5,
      "s_time": 1735515845,
      "e_time": 1735515845,
      "w_data2": 3601
    },
    {
      "pv_sn": "220702xxxxxxxx",
      "code": 45058,
      "num": 6,
      "s_time": 1735512304,
      "e_time": 1735512304,
      "w_data1": 65535,
      "w_data2": 61935
    },
    {
      "pv_sn": "220702xxxxxxxx",
      "code": 45058,
      "num": 7,
      "s_time": 1735512350,
      "e_time": 1735512350,
      "w_data2": 3601
    },
    {
      "pv_sn": "220702xxxxxxxx",
      "code": 45058,
      "num": 8,
      "s_time": 1735512365,
      "e_time": 1735512365,
      "w_data1": 65535,
      "w_data2": 61935
    }
  ]
}
```

Also, the `RealDataNewReq` messages are containing codes I would like to understand:
```
2024-12-30 09:21:10.661725 | DTU -> CLOUD | command 220D (RealDataNewReq) | seq 005D | {
  "sn": "4143A0xxxxxx",
  "time": -1,
  "ap": 1,
  "ver": 1,
  "sgs_datas": [
    {
      "sn": "220702xxxxxxxx",
      "v": 2279,
      "freq": 4999,
      "p": 1423,
      "q": 38,
      "i": 65,
      "pf": 999,
      "temp": 39,
      "wnum": 1,
      "link": 1,
      "mi_signal": 11272300
    }
  ],
  "pv_datas": [
    {
      "sn": "220702xxxxxxxx",
      "pi": 1,
      "v": 406,
      "i": 178,
      "p": 722,
      "et": 22142,
      "ed": 23,
      "code": 50921472    <-----
    },
    {
      "sn": "220702xxxxxxxx",
      "pi": 2,
      "v": 406,
      "i": 192,
      "p": 783,
      "et": 22151,
      "ed": 24,
      "code": 50331648    <-----
    }
  ]
}
```

**Describe the solution you'd like**
Is there a a table of warning codes available somewhere?

**Describe alternatives you've considered**
Checked over GitHub without success. I'll try to check inside the Smiles app code soon...

**Additional context**
- Sorry for creating an issue I would have started a "discussion" but this is not enabled on this repo.
- Using the smiles installer app and connecting using the toolkit doesn't report any error, neither warnings.

At the beginning of my experimentation, I also had some unknown messages of code `2209`. Any insight to decode it?
```
2024-12-28 17:08:43.140417 | DTU -> CLOUD | command 2209 (???) | seq 00BD | see /tmp/hoymiles.local5.pcap_msg_2209_29

$ xxd /tmp/hoymiles.local5.pcap_msg_2209_29
00000000: 0a0c 3431 3433 4130 7878 7878 7878 10a2  ..4143A0xxxxxx..
00000010: c5c0 bb06 1801 3210 0884 c7e3 80ba a810  ......2.........
00000020: 1036 289d 8502 3009                      .6(...0.
```

Thanks a lot for your help and for your great work!

suaveolent, 2025-01-03:
Unfortunately I have not yet found the corresponding  (if there even is one) mapping the application or implemented the WarnReq but this is definately something which could be added in the future.

I'll keep the issue open as a todo.

Sonnenwerk, 2025-07-06:
I have identified those error codes:

[image]

My assumptions:
589824 = running but producing under minimum
50986752 = running normal
134872832 = overcurrent ?
0 = daily reset ?

a. Update intervall in hoymiles webapp is 5 min. I don´t know the intervall in ha.
b. The logs do not match, because i can not see the errorcodes in webapp.