suaveolent/hoymiles-wifi #49: a temperature reading of 0 is missing with get-real-data-new

rscheff, 2025-01-01:
These days, the internal temperature of my HMS-800W-T2 frequently transitions throught 0 °C over the day.

However, the output of "get-real-data-new" does not show the temperature line at all.

Is that field not decoded when it is zero (which shouldn't happen with protobuf), or is there a bug in the Hoymiles Firmware (V01.03.09)?

To reproduce:
Repeatedly read out the values of a Microinverter, while slowly cooling it, or warming it through 0 °C. (e.g. put it into the freezer with constant voltage source). 

Here are two polls, one with -0.3°C and the other one with 0 °C - the shorter packet seems to indicate that the TLV is actually missing from the DTU/Microinverter:

0°C:
0030   -- -- -- -- -- -- 48 4d a3 11 00 01 49 56 00 29   r.+...HM....IV.)
0040   0a 13 32 30 32 35 2d 30 31 2d 30 31 20 31 34 3a   ..2025-01-01 14:
0050   35 38 3a 34 33 20 80 e1 01 28 93 94 d5 bb 06      58:43 ...(.....

0030   -- -- -- -- -- -- 48 4d a2 11 00 01 ad ee 00 7d   ".@...HM.......}
0040   0a 0c 34 31 34 33 39 32 33 38 32 30 30 39 10 9c   ..414392382009..
0050   94 d5 bb 06 18 01 28 01 4a 21 08 89 c0 e0 91 a9   ......(.J!......
0060   82 05 10 01 18 99 12 20 86 27 28 c4 01 38 08 40   ....... .'(..8.@
0070   e8 07 50 06 60 01 a0 01 cd 80 14 5a 17 08 89 c0   ..P.`......Z....
0080   e0 91 a9 82 05 10 01 18 c1 02 20 20 28 67 30 d4   ..........  (g0.
0090   8d 11 38 79 5a 17 08 89 c0 e0 91 a9 82 05 10 02   ..8yZ...........
00a0   18 c9 02 20 1f 28 68 30 f1 f4 05 38 66 60 c4 01   ... .(h0...8f`..
00b0   68 df 01                                          h..

-0.3°C:

0030   -- -- -- -- -- -- 48 4d a3 11 00 01 49 56 00 29   r.^...HM....IV.)
0040   0a 13 32 30 32 35 2d 30 31 2d 30 31 20 31 34 3a   ..2025-01-01 14:
0050   35 38 3a 34 33 20 80 e1 01 28 93 94 d5 bb 06      58:43 ...(.....

0030   -- -- -- -- -- -- 48 4d a2 11 00 01 e5 f9 00 86   ".K...HM........
0040   0a 0c 34 31 34 33 39 33 31 33 36 33 30 32 10 9f   ..414393136302..
0050   94 d5 bb 06 18 01 28 01 4a 2c 08 82 c6 cd 98 a9   ......(.J,......
0060   82 05 10 01 18 8b 12 20 86 27 28 66 38 04 40 e8   ....... .'(f8.@.
0070   07 48 fd ff ff ff ff ff ff ff ff 01 50 05 60 01   .H..........P.`.
0080   a0 01 d8 80 c8 07 5a 17 08 82 c6 cd 98 a9 82 05   ......Z.........
0090   10 01 18 d0 02 20 18 28 52 30 c5 dd 17 38 60 5a   ..... .(R0...8`Z
00a0   17 08 82 c6 cd 98 a9 82 05 10 02 18 bd 02 20 08   .............. .
00b0   28 19 30 cb d7 08 38 1d 60 66 68 7d               (.0...8.`fh}

suaveolent, 2025-01-03:
Hm thanks, I could not reproduce it with my inverter. I, however, still did not upgrade to `V01.03.09` since I want to record the update process in the hope that we could trigger it locally in the future.

rscheff, 2025-02-03:
Could it be that whatever you are using to graph these values is implicitly zeroing undefined values?

Because looking at the packet capture of an older version (Hoymiles downgraded one of my HMS), zero values (regardless of temperature, amps, voltage, accumulated energy, ...) are omitted - presumably the software they use is "optimizing" the TLV values sent by not sending zero values at all.

My workaround here was to improve my scripting to use JSON directly, and initializing the values to zero IFF a response comes back, and leaving UNDEF when there is no response/timeout.

suaveolent, 2025-02-14:
Yes, I am using the Home Assistant integration I wrote (https://github.com/suaveolent/ha-hoymiles-wifi).

The plot is done via the integrated tools of Home Assistant. Values are reported directly. 
If however, the value is missing or the inverter is offline I am simply returning 0 for this value. Which basically matches with your  approach. 

I guess this issue can be closed then?

rscheff, 2025-02-14:
Yes - it's the HMS inverters who omit the values in their TLV data - formally correct would be to declare them "U"navailable  / NaN / missing. However when the other data of the TLV arrives, it's a save enough assumption these missing values all to be zero.