suaveolent/hoymiles-wifi #56: Warning message about new version of Protobuf 6

OpenHelios, 2025-03-30:
Currently there is a warning message about new version of Protobuf 6.

Steps to reproduce the behavior:
execute hoymiles-wifi

**Current** output:

```
/usr/lib/python3.13/site-packages/google/protobuf/runtime_version.py:98: UserWarning: Protobuf gencode version 5.29.3 is exactly one major version older than the runtime version 6.30.1 at AppGetHistPower.proto. Please update the gencode to avoid compatibility violations in the next runtime release.
```

**Expected** output: no warning

suaveolent, 2025-03-31:
Thanks I’ll have a look at it.

uklatt, 2025-04-20:
You can to downgrade protobuf to version 5.29.3:
pip install protobuf==5.29.3

Tschau
Uwe

nebucatnezr, 2025-05-03:
> You can try to downgrade protobuf to version 5.29.3: pip install protobuf==5.29.3
> 
> Tschau Uwe

Hey, tried it with your reinstall of protobux downgrade and now it says "1 file and 0 cells to analyze" - for minutes?

Wookbert, 2025-06-04:
Note that the latest [HA Core Beta Release Notes](https://github.com/home-assistant/core/releases/tag/2025.6.0b5) reveals that HA will from now on come with protobuf 6.x:
[image]

This means that this issue must be most definitely resolved by the maintainer @suaveolent or someone else.

suaveolent, 2025-06-04:
Thanks,  I will have more time in June to work on the project and will fix the issues then.

Sonnenwerk, 2025-06-09:
> Thanks, I will have more time in June to work on the project and will fix the issues then.

Looging forward to it ;-)

Wookbert, 2025-06-26:
@suaveolent Any news on this? I’m currently holding back any Home Assistant updates, as it otherwise would break hoymiles-wifi.

closed 2025-07-01 by suaveolent

Wookbert, 2025-07-01:
@suaveolent So new, updated release including this fix when and where? Or am I too blind!? 😎 [Here](https://github.com/suaveolent/hoymiles-wifi/releases) it still says Feb 28, 2025!?

PR #66 (2025-07-01): Add support for controlling hybrid inverters
Add support for controlling hybrid inverters
Fixes #56