suaveolent/hoymiles-wifi #60: Regular disconnects

dm807cam, 2025-05-13:
**Describe the bug**
HA is regularly disconnected from hoymiles inverter resulting in a lot of missing data. I tried to adjust the query time but it doesn't seem to have any affect on this. The official Hoymiles App (S-Miles) does not show a similar behavior which leads me to think that it is a bug with the HA add-on.

**Screenshots**

[image]

suaveolent, 2025-05-13:
The integration pulls data way more often than the cloud update does. 

Could it be that the inverter is regularly losing the WiFi connection?

dm807cam, 2025-05-16:
seems that the issue really originates from the inverter that keeps disconnecting and reconnecting from my network. I contacted the support and fixed the issue with a custom template that filters 0s in the meantime.

closed 2025-05-16 by dm807cam