suaveolent/hoymiles-wifi #72: --disable-interactive doesn't work with set-power-limit and --power-limit <n>

rscheff, 2025-08-11:
**Describe the bug**
I am trying to implement a net-zero control (no energy injected into the grid), using a RasPi and hoymiles-wifi plus a bunch of other, not integrated pieces.

In order to restrict how much power the microcoverter is generating, I want to adjust the power-limit quite frequently (probably in the order of 1-10 sec), and tried this with the CLI execution of 

**To Reproduce**

hoymiles-wifi --host <x.x.x.x> set-power-limit --disabled-interactive --power-limit 20

But then an interactive prompt comes back, asking if I'm sure, asking again about the power limit (not using the value provided on the CLI), and asking once more if that is the setting to apply...

According to the manual, with --power-limit <n> the question about what power-limit to apply should be skipped, and with --disabled-interactive, the double-safety prompts should be disabled. But both these options are apparently without function...

**Expected behavior**

The documented functionality of these two extended parameters should work as expected...

**Desktop (please complete the following information):**
 - OS: RasPi Bookwork (Debian Based)

rscheff, 2025-08-11:
Apparently, the async_set_power_limit code actually runs twice when running form the CLI; once with the parameters given as arguments, and subsequently in an interactive way...

suaveolent, 2025-08-11:
@rscheff the flag should be `--disable-interactive` not `--disabled-interactive`

Also note that the feature should not be used for zero feed in due to the way hoymiles handles writes to the EEPROM. The feature changes the limit persistently which if used excessively might destroy your device. 
I will add a note the the Readme too.

suaveolent, 2025-08-11:
@rscheff Thanks. It's fixed with `v0.5.1` now.

Regarding usage of EEPROM: I don't know exactly what is being used. Some users reported destroying their inverter after using the power limit for zero feed in.

Unfortunately, there is no alternative method unless hoymiles provides one.

rscheff, 2025-08-11:
Thanks for this very rapid response and fix!