suaveolent/hoymiles-wifi #75: Encrypted commands do not work since firmware V01.01.00 on DTU-WLite-S

intek-ck, 2025-08-14:
After the firmware update V01.01.00 encrypted communciation does not work. Unencrypted commands (e.g. app-information-data) do work. If no or wrong enc_rand is used, DTU does not answer at all. If correct enc_rand (from DTU) is used, socket seems to be immediately closed by the DTU with no answer. I will do further investigation.

kumy, 2025-08-30:
I'm with you, such practice is not acceptable. I'm glad I didn't updated yet.

nename0, 2026-09-23:
For me the problem was that my DTU-WLite-S has a wrong internal clock (somwhere in 2022). Since I did not connect it to the internet at all.
The library always send the current time in requests. Apparently the DTU will refuse to answer if this time is to far off it's own clock. I added a workaround for this in #111

suaveolent, 2026-10-27:
Merged that PR, thanks for the dfeedback.

closed 2026-10-27 by suaveolent