suaveolent/hoymiles-wifi #78: inverter serial number in HEX

btgh, 2025-09-13:
**Describe the bug**
The command line option —inverter-serial-number will not accept a hexadecimal number but my inverter serial number are all in hex 

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

macrl2000, 2025-09-16:
there seems to be a "device_serial_number" which is hex and then there is a "serial_number" which is integer. you might need this one.

maybe try a small python script like below to get a datadump

```python
import asyncio
from hoymiles_wifi.dtu import DTU

dtu_ip = "192.168.7.11"

# main loop
async def main():

    print(f"Connecting to DTU")
    dtu = DTU(dtu_ip)
    nwi = await dtu.async_network_info()

    if nwi:
        print(f"Found DTU: {nwi.dtu_sn}")
    else:
        print(f"Unable to get DTUresponse, exiting ...")
        exit()

    rdata = await dtu.async_get_real_data_new()
    if rdata:
       print(rdata)

loop = asyncio.new_event_loop()
asyncio.set_event_loop(loop)

loop.run_until_complete(main())
loop.close()
```

suaveolent, 2025-09-16:
@btgh  you should be able to get the serial number as int by running the `get-gateway-info` command first.

closed 2025-10-31 by github-actions[bot]